MarlinFirmware/Marlin #10612: Use E1 Port for Z2

WillShawx, 2018-05-03:
### Description

Hi, i am using a MKS Gen V1.4 On a Reprap printer with dual Z Motors.

Is it possible for me to use the spare E1 Port to drive the second Z Motor by making some firmware modifications or do i have to use an adaptor like this (https://www.ebay.co.uk/itm/Stepper-motor-adapter-to-put-2-stepper-motors-in-series-not-parallel-z-axis-3D/282873720059?hash=item41dc966cfb:g:-NEAAOSwhcNadNCl)

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-05-03:
If you simply enable `Z_DUAL_STEPPER_DRIVERS` the first unused E plug will be automatically assigned to Z2.

WillShawx, 2018-05-03:
So how does the board differentiate between the extruder motor and the Z Motor? do i have to boot it up without Z2 Plugged in first? or is it simply plug and play once i upload the new sketch? 

thanks

thinkyhead, 2018-05-03:
Plug and play. The stepper driver at the position with "E1" silkscreened on the board becomes your Z2 stepper driver.

closed 2018-05-03 by thinkyhead

WillShawx, 2018-05-03:
Thanks! Much Appreciated