MarlinFirmware/Marlin #10613: Can't compile bugfix 2.0

Iamturbo1978, 2018-05-03:
### Description

[Iamturbo1978_2018.05.03.zip](https://github.com/MarlinFirmware/Marlin/files/1972592/Iamturbo1978_2018.05.03.zip)

teemuatlut, 2018-05-03:
If using Arduino 1.9 beta didn't work (it should have), add this to your IDE preferences.txt:
`build.path=C:\arduino_builds`

Iamturbo1978, 2018-05-03:
> Marlin 2.0 is not released so the majority of users know that they are beta testers and comfortable with the consequences of being on the bleeding edge.

That's the main headache for me, I want to be part of the bleeding edge... Mostly because of the better planer, and linear advance 1.5

> The rest of the issues are in `pins_AZTEEG_X3_PRO.h` and are due to another issue with Arduino IDE, comments on the same line as a define cause a problem with token concatenation.

That makes it a big issue, I added comments for everything. That way I don't have to remember what I changed, and what the original value was. That's a lot of comments to delete

> If using Arduino 1.9 beta didn't work (it should have),
add this to your %localappdata%\Arduino15\preferences.txt:
build.path=C:\arduino_builds

I tried your suggestion, and still no go. I would add `build.path=C:\arduino_builds` restart the Arduino IDE, try to compile, it fails, I check the preferences.txt file, and `build.path=C:\arduino_builds` is missing.

> Please consider dumping the Arduino compiler and switching to PlatformIO.

The latest WIKI has a write up on using PlatformIO. Once setup it's only 2 or 3 mouse clicks to upload your board.

I dont think I have any other choice but to start using PlatformIO. I did not know there was a WIKI that went in that much detail. I'll have to give it a shot.

p3p, 2018-05-03:
> I dont think I have any other choice but to start using PlatformIO.

Other than the comment issue (admittedly annoying) there is no reason you can't use Arduino IDE 1.9 beta, I did manage to compile with your configs successfully.

Iamturbo1978, 2018-05-03:
> Please consider dumping the Arduino compiler and switching to PlatformIO.

The latest WIKI has a write up on using PlatformIO. Once setup it's only 2 or 3 mouse clicks to upload your board.

I'll have to give it a shot. I'll reply to this post if it works or not

cjsoong, 2018-05-04:
@Iamturbo1978 
arduino ide 1.85;  I add this line "build.path=c:\tmp"
[image]

thinkyhead, 2018-05-04:
Also note that, apart from a few features, the `bugfix-1.1.x` branch has full parity with `bugfix-2.0.x`. So unless you want to use the `BEZIER_JERK_CONTROL` option, you can stick with `bugfix-1.1.x` for a while longer.

rsmith21, 2018-05-10:
I'm not very experienced so please don't be cross, one solution ( work around) I have found to the file name to long error is to create a TMP folder on the root of C:\ and edit the system advanced variables to point at that folder, reducing the path to tmp while compiling I think, anyhoo it worked for me after a restart.

thinkyhead, 2018-05-10:
> create a TMP folder on the root of C:\ and edit the system advanced variables to point at that folder

Arduino 1.9 beta is the option we'll continue to recommend, especially as many are using macOS and Linux, where setting system-wide environment variables requires some expertise.

closed 2018-05-10 by thinkyhead