MarlinFirmware/Marlin #10616: Arduino crash after long Z homing with Delta

Nossolov, 2018-05-04:
Hello.
I am building a big delta printer, with arduino mega 2560, a Ramps 1.4 and Marlin 1.1.x. If i ask for a Z homing, everything works fine for less than 400mm travel. If i try the same with a travel longer than 400mm, all motors stop at the same time after a while, arduino stop working, the green led on ramps blink fast, Reset switch doesn't work and i must disconnect USB cable and switch off the power to restart.
I have 400 steps motors and 32 microsteps. May be it's too much for the arduino, and homing at fast speed is not possible ? May be there is a security in firmware when homing take too much time ?

thinkyhead, 2018-05-04:
Correct. 32x microsteps is too much for an AVR to keep up with the requested feedrates. You'll have to reduce your maximum feedrate to prevent the board from rebooting.

Nossolov, 2018-05-09:
Some test made :
Feed rate of Z homing reduce from 200 x 60 to 50 x 60 : works fine. Now, G33 command works too.
Increase Z homing feedrate little by little, up to 80 x 60 without pb.
Thank's a lot !
I suppose that i have to use Marlin 2.0 with a MKSbase 32 bits board ?

thinkyhead, 2018-05-10:
> I suppose that i have to use Marlin 2.0 with a MKSbase 32 bits board?

Yes.

closed 2018-05-10 by thinkyhead