MarlinFirmware/Marlin #10617: DRV8825's

WillShawx, 2018-05-04:
### Description

When i attempt to use DRV8825's on Marlin 1.1.8, the motor's vibrate and don't spin. However, using the exact same drivers on Repitier Firmware, i get no issues at all.

Nossolov, 2018-05-04:
I change A4988 by DRV8825 on a Ramps board with Marlin 1.1.8. No problem but need to adjust current.

WillShawx, 2018-05-04:
I’m using MKS GEN 1.4

p3p, 2018-05-05:
Marlin 1.1.8 defaults to '#define MINIMUM_STEPPER_PULSE 0' in `Configuration_adv.h` (this was changed in bugfix) DRV8825's require that you change it to '#define MINIMUM_STEPPER_PULSE 2', you didn't supply configs as the issue template requested so I can only assume you haven't.

thinkyhead, 2018-05-08:
> DRV8825's require that you change it to #define MINIMUM_STEPPER_PULSE 2

It may not matter on AVR, but it can't hurt.

closed 2019-05-16 by thinkyhead