MarlinFirmware/Marlin #10618: [bugfix-2.0.x] RE-ARM - Layer Fan et  E0_AUTO_FAN_PIN

J-charles-C, 2018-05-04:
### Description

On MARLIN 2.0 i can't activate D6 et D5 for layer Fan et Extruder Fan

I have a RE-ARM With RAMPS 1.6.

I have activate there fan on my MKS GEN and Marlin 1.1.8 with success but not with RE-ARM.

I begin by  E0_AUTO_FAN_PIN

I have coded 👍 #define E0_AUTO_FAN_PIN 5

No Success. Fan is off when extruder is hot.

best

benlye, 2018-05-04:
Related to #10599?  What does M43 report for the pin?

J-charles-C, 2018-05-04:
I have forgotten,

I have two extruder and my board is in EEB.

best

J-charles-C, 2018-05-05:
With IFR520 connect to Servo Pin 4, Layer Fan is OK.

With same wiring on Servo Pin 5, extruder fan is On, all the time.

best

kAdonis, 2018-05-05:
you probably use the RAMPS Servo header to connect the extruder fan mosfet. When the RAMPS is connected to an Arduino Mega the corresponding Arduino Pins are D11,D6,D5 and D4.
the Re-Arm has a completely different pin numbering scheme.
Take a look at pins_RAMPS_RE_ARM.h:

```
// Servos
//
#define SERVO0_PIN         P1_20   // (11)
#define SERVO1_PIN         P1_21   // ( 6) also on J5-1
#define SERVO2_PIN         P1_19   // ( 5)
#define SERVO3_PIN         P1_18   // ( 4) 5V output
```
so you need to define in Configuration_adv:

`#define E0_AUTO_FAN_PIN   P1_18`

to use Servo3 Pin (which was D4 on Arduino) for the extruder fan

J-charles-C, 2018-05-06:
I,

If  i do that, i have a SPIO erreur at compilation.

best

kAdonis, 2018-05-06:
please post the  compiler output from PlatformIO

J-charles-C, 2018-05-06:
I,

`arning! Library `{'requirements': None, 'name': 'CMSIS-LPC1768'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': None, 'name': 'CMSIS-LPC1768'}` is a built-in library (included in framework,SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: NXP LPC > NXP mbed LPC1768
SYSTEM: LPC1768 100MHz 64KB RAM (512KB Flash)
DEBUG: CURRENT(custom) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(light)
Collected 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CMSIS-LPC1768> v0.0.0
|-- <U8glib-HAL> v0.4
Calculating size .pioenvs/LPC1768_debug_and_upload/firmware.elf
text       data     bss     dec     hex filename
231204     1388   14028  246620   3c35c .pioenvs/LPC1768_debug_and_upload/firmware.elf
================================================= [SUCCESS] Took 3.80 seconds =================================================

========================================================== [SUMMARY] ==========================================================
Environment megaatmega2560              [SKIP]
Environment megaatmega1280              [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [SKIP]
Environment LPC1768_debug_and_upload    [SUCCESS]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [SKIP]
Environment teensy20                    [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]`

Config : PIO Build LPC1768_debug_and_upload

pins_RAMPS_RE_ARM.h
`/**
 * Marlin 3D Printer Firmware
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 * Copyright (C) 2017 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * Re-ARM with RAMPS v1.4 pin assignments
 *
 * Applies to the following boards:
 *
[... 360 lines cut ...]

/**
 *  The following mega2560 pins are NOT available in a Re-ARM system
 *  7
 *  17
 *  22
 *  23
 *  25
 *  27
 *  29
 *  32
 *  39
 *  40
 *  42
 *  43
 *  44
 *  45
 *  47
 *  64
 *  65
 *  66
 */
`

Thanks for your help.

darkusone, 2018-05-06:
Same issue here, i have an EFB, i installed an autolevel endstop this weekend, but now, layer fan is allways on and extruder fan never works... 

#define E0_AUTO_FAN_PIN 9
#define E1_AUTO_FAN_PIN -1
#define E2_AUTO_FAN_PIN -1
#define E3_AUTO_FAN_PIN -1
#define E4_AUTO_FAN_PIN -1
#define EXTRUDER_AUTO_FAN_TEMPERATURE 50
#define EXTRUDER_AUTO_FAN_SPEED   240  // 255 == full speed

But if i set "FAN ON" by octoprint, it works normally

any idea?

kAdonis, 2018-05-06:
@J-charles-C  you are welcome
@darkusone You have a different Problem
I assume you are using an Arduino Mega and no Re-Arm?
On a RAMPS EFB pin 9 is used for the parts cooling fan (that's what you switch with FAN ON), you need a different pin for your extruder fan, pins 6,5 or 4 (on the RAMPS Servo header) are often used and you need an Fan-Extender or mosfet to control your Extruder Fan.

darkusone, 2018-05-07:
oops... i'm sorry then... but with marlin 1.1.0 i have no problem with it, works perfect, were can I find information about?

kAdonis, 2018-05-07:
look at the Configs and pins_RAMPS.h from Marlin1.1.0 you used before, if you have them.
Information about wiring a Hotend Fan are here:
http://www.instructables.com/id/Configuring-and-using-Reprap-Ramps14-RRD-Fan-Exten/

darkusone, 2018-05-07:
Thanks! your best and sorry for confused thread...

J-charles-C, 2018-05-08:
I, 

My problem is not solved.
best

J-charles-C, 2018-05-08:
I,

My problem is not solved !

My fan extruder is always on.

best

> Le 8 mai 2018 à 17:20, Scott Lahteine <notifications@github.com> a écrit :
> 
> Closed #10618 <https://github.com/MarlinFirmware/Marlin/issues/10618>.

thinkyhead, 2018-05-10:
@J-charles-C — Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as requested in the Issue template) and drop them into your next reply. We'll check them over and see if anything is amiss.

J-charles-C, 2018-05-14:
I,

Sorry, i don't have internet in travel.

The zip is on the first line of post in "archive.zip"

best

J-charles-C, 2018-05-16:
I 
kAdonis solved my problem.

Soluce : #define E0_AUTO_FAN_PIN   P1_18

But firmware haven't error on  compilation only if i use LPC1768_debug

Thanks boys.

Jc

closed 2018-05-16 by J-charles-C