MarlinFirmware/Marlin #10621: tmc2130 Fast Movement Noises on delta

Kyzuuh, 2018-05-05:
### Description

tmc 2130's produce very loud grinding noise on fast movements.  only seems to happen on quick movements in UBL or auto calibrate.  moving quickly via pronterface does not seem to reproduce the noise, but I may nopt be moving it quickly enough to produce the sound.  

This happens in spreadcycle and stealthchop modes.  Current values have been toyed with extensively, not changing the sound at all.  I'm on marlin 2.0 on an Arduino Mega 2560 and a RAMPS 1.4 
The version is the latest.  I'm using software SPI with custom defined pins (I'll include my pins file)
Sensorless homing also seems to have an issue where all axis do not rise before homing individual axis.  When my current values were very low, this issue went away, not sure why.  

This noise is violent and triggers my piezo probes underneath my bed, so bed levelling isn't possible for me until I figure this out. Tried posting on the forums but to no avail.  Will be trying 1.1.8 soon to see if it changes anything.  

### Steps to Reproduce

Choose UBL cold mesh or auto calibrate, wait for fast movements, enjoy the painful sounds.

**Expected behavior:** 
Normal stuff, normal sounds and quiet movements

**Actual behavior:** 
On fast movements, there is some horrible noise coming from the motors.  seems to only happen on fast movements inside auto calibrate or UBL.  moving z quickly through pronterface does not seem to create the sound.  

[stuff.zip](https://github.com/MarlinFirmware/Marlin/files/1976235/stuff.zip)

ikarisan, 2018-05-07:
Maybe same problem (cause) as mine?!
https://www.youtube.com/watch?v=f19ILbUh5SQ

Are you using the bugfix-2.0.x version?

Kyzuuh, 2018-05-07:
@ikarisan it does sound somewhat similar.  You're also using 2130's? 

I'm on the latest 2.0 bugfix, yes.

boelle, 2019-02-20:
@Kyzuuh what voltage? you need to be at least 24V before noise is gone or almost gone

closed 2019-05-16 by thinkyhead