MarlinFirmware/Marlin #10622: XYZ on display keeps switches between ??? and XYZ

ProtoPrt, 2018-05-05:
My current display on anet a8 has the x,y and z flashing between the letters and ?. This started after I flashed he firmware to Marlin after using Marlin-config. If I go into the menu I can get to the screen with Prepare, Control... If I select one of those items it goes back to the main screen. Any help on how to resolve this?
Marlin 1.1.8

gloomyandy, 2018-05-05:
This is normal, the flashing X/Y/Z/? indicates that the current position of the axis is unknown as you have not yet homed it. Once you have performed a home it will stop flashing.

ProtoPrt, 2018-05-05:
Printer won’t home. I’ve sent G28 in the console and the menus on the printer won’t respond.

gloomyandy, 2018-05-05:
Again I'm pretty sure that what you are seeing is normal, when you disconnect I think the controller will reset, so again it will not know the positions of the axis. The same thing is true when you power off and on. So I don't think there is a problem with the flashing items on the display and you should probably close this issue

If you are having other problems like the menus not working then you should create a new issue for the actual problems you have. Be sure to fill in all of the information required (like supplying your config files and describing exactly what you are doing and what happens).

ProtoPrt, 2018-05-05:
Thank you gloomyandy. The menu problem was because they moved the okay button to the center now instead of the far right. Everything works as it should now.

closed 2018-05-05 by ProtoPrt