MarlinFirmware/Marlin #10624: [Q] using 2nd extruder as heated chamber control/monitor on LCD

boelle, 2018-05-05:
i have asked this on facebook too but want target as wide i can

i have a pretty good setup and only need to nail the filament change/runout

while i added a small pigtail for max chips so i can use thermocouples i have also attached a thermistor to T2 so i can monitor the chamber temp on the LCD

my plan was to enable a 2nd extruder and use that to monitor and maybe later control the temp

that has some downsides thou:

i use the 2nd extruder stepper on the ramps board as i have a dual z stepper and endstop setup to keep the 2 Z in sync automatic

D8-D10 are taken up also and i would really hate to making any more mod to the wiring. to control heating of the chamber my plan was to use a free pin on aux 1 or 2

any tips on how i could make a workarround would be much welcomed

current config and pins file in the zip

pins file is changed as the default pin for the filament runout switch was taken up
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1976477/configuration.zip)

closed 2018-05-11 by boelle

thinkyhead, 2018-05-13:
> does the idea sound good? or are there better ways?

As a thoroughgoing software engineer, I believe code written
specifically for the application is better than hacking the system.

boelle, 2018-05-13:
i was also more thinking of it as workaround until the feature is there