MarlinFirmware/Marlin #10627: Auto calibration fails to reach bed starts at endstops

bssman89, 2018-05-05:
### Description

The Auto Calibration fails to reach the heated bed of my printer and starts at the top near the endstops
the delta height is correct at 315 mm not sure if a bug in the firmware or just an issue on my part 

im using the firmware release Marlin-AC-G-force8a-AC  on a flsun delta printer 

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

forkoz, 2018-05-06:
That's G-force... not plain marlin

bssman89, 2018-05-07:
i fixed it i had some settings wrong in the firmware

closed 2018-05-07 by bssman89