MarlinFirmware/Marlin #10629: [2.0.x] TMC2130 X not moving on SpreadCycle

G-Pereira, 2018-05-06:
### Description
When I use StealthChop all axis work well but because I want to try higher speeds i've changed to SpreadCycle but X axis doesn't move at all. I've changed back and forth between the two modes and in StealthChop the X move nicely but never on SpreadCycle.

vilsed, 2018-05-06:
What kind of drivers are you using? Aliexpress chinesium ones require couple of little solder bridges to have spreadCycle enabled, as per http://marlinfw.org/docs/features/tmc_drivers.html#troubleshooting

I was having the same problem, just noticed it a couple of days ago.

G-Pereira, 2018-05-07:
Hi, I had those problems with the chinese ones, so I am using others from a trusted manufacturer close from my home country: https://www.staticboards.es/productos/tmc2130sb-spi/
This ones have those bridges connected with 0 ohm resistors.
So the problem it's not from that, any ideas?

G-Pereira, 2018-05-20:
The X driver was damaged

closed 2018-05-20 by G-Pereira

comps, 2018-05-30:
I'll just add that this exact thing has happened to me as well. The motor would just enable and do one or two steps in the correct direction, then stop (coils enabled). For example homing would do the first 1-2 steps, then lock up, and another stop + 1-2 steps for the homing bump.

M122 reports no stall, as far as the driver is concerned, the motor is working just fine.

Switching to stealthChop "fixed" it, but `SENSORLESS_HOMING` couldn't be obviously used.

I'll try to get another driver, ... I'm just pointing out here that the issue is not that uncommon, maybe a bad batch. Checking under microscope, there don't seem to be any obvious shorts on the PCB, nor damaged traces. The IC soldering is a bit shoddy (Fysetc drivers), but no faults.

knoopx, 2018-07-13:
I just switched to SpreadCycle (to workaround layer shifting) and exactly the same thing happened to me. My X stepper was not moving at all but replacing it with another one made it move again. I'm also using the FYSETC ones. Maybe something is "breaking" the X driver? It's is strange so much coincidence.

Delorean14, 2019-02-14:
I am having a similar problem, only it doesn't matter whether or not StealthChop or SpreadCycle are enabled.  The X axis will move one to two steps and lock, almost like it's hit the endstop.  I have sensitivity set to 8.  The X axis is a single motor carriage.  My Y axis is working perfectly and it is a dual motor gantry.