MarlinFirmware/Marlin #10635: [2.0.x] compile warning (redefinition)

GMagician, 2018-05-06:
Don't know where L6470.h is used but it define DEC as 0x06 but DEC is already defined in MarlinSerial.h as 10.
This gives me a compile warning but I don't know how dangerous this can be

boelle, 2018-05-06:
you are using IDE 1.90 right?

GMagician, 2018-05-06:
visualstudio and platformio

boelle, 2018-05-06:
just a thought :-/

p3p, 2018-05-06:
DEC is defined in the Arduino print.h header as 10, so it can be assumed to be that value within the Arduino framework, L6470.h is part if an external library and at least the version downloaded automatically by platformio does not redefine DEC, it does have `#define L6470_DEC 0x06` but that isn't a problem. When submitting compile issues be sure to supply the generated error message (and context) we can infer a lot more information from that.

GMagician, 2018-05-06:
> L6470.h is part if an external library and at least the version downloaded automatically by platformio does not redefine DEC, it does have #define L6470_DEC 0x06

I have two L6470.h in .piolibdeps folder, one in 'Arduino-L6470' and other in 'Arduino-L6470@src-569f877a8e29bbf95e5e0e1923f96a71'
It seems that my environment compile using second folder, not first. Here is why I have such warning.
In second folder is really defined DEC

GMagician, 2018-05-06:
@p3p thanks to your hint I cleaned up .piolibdeps folder and now everything is ok

closed 2018-05-06 by GMagician

p3p, 2018-05-06:
I just looked and it appears thinkyhead committed a fix for the badly names defines in the lib a few months ago.