MarlinFirmware/Marlin #10636: [2.0.x] compile warning (narrowing conversion)

GMagician, 2018-05-06:
In file 'Adafruit_NeoPixel.cpp' there are two point without cast that generate more warnings.
`G8668B7F5 narrowing conversion of 'xxxx' from 'double' to 'const uint8_t {aka const unsigned char}' inside { } [-Wnarrowing]	Marlin	C:\Users\Giuliano\Documents\GitHub\Marlin\.piolibdeps\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp`

cast (uint_8) is missing in _G1_ and _S1_ defines but this file is not part of marlin, it is in .piolibsdeps.
I don't know who has to fix it...

p3p, 2018-05-06:
We do not maintain the Ardafruit Neopixel library, if the library functions as expected we have to assume their devs decided it was fine.

GMagician, 2018-05-06:
@p3p thanks to your hint in #10635 I cleaned up .piolibdeps folder and now everything is ok

closed 2018-05-06 by GMagician