MarlinFirmware/Marlin #10638: Arc moves (G2/G3) and bezier curves (G5) don't apply leveling correction with non-delta UBL

ManuelMcLure, 2018-05-06:
### Description

Arc moves (G2/G3) and bezier curves (G5) don't appear to apply leveling correction with UBL on anything but a delta.

Found through code inspection.

It looks like `plan_arc()` is calling `planner.buffer_line_kinematic()` directly and  `planner.buffer_line_kinematic()` does not apply any leveling unless `PLANNER_LEVELING` is set. `PLANNER_LEVELING` is defined as
```
#define UBL_SEGMENTED  (ENABLED(AUTO_BED_LEVELING_UBL) && (ENABLED(DELTA)))
#define ABL_PLANAR     (ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_3POINT))
#define ABL_GRID       (ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR))
#define OLDSCHOOL_ABL  (ABL_PLANAR || ABL_GRID)
...
#define PLANNER_LEVELING      (OLDSCHOOL_ABL || ENABLED(MESH_BED_LEVELING) || UBL_SEGMENTED || ENABLED(SKEW_CORRECTION))
```
so `PLANNER_LEVELING` is **not** set if using UBL on anything but a delta and therefore it looks like leveling is not applied to arc moves.

The same applies to `cubic_b_spline()` and bezier curves.

thinkyhead, 2018-05-08:
Patches submitted, and will be merged after review.

boelle, 2019-02-20:
@ManuelMcLure is this still an issue using bugfix 2.0?

closed 2019-02-20 by thinkyhead