MarlinFirmware/Marlin #10639: [FR] TMC2208 sensorless home using StallGuard2

viper93458, 2018-05-07:
**### Description**
Looking for the ability to use sensorless homing with the TMC2208 drivers which is currently only available on the TMC2130's via configuration_adv.h

Per the data sheet, it looks as if the 2208 has the ability, but software support is needed. Perhaps this is already in development?

**Expected behavior:** [What you expect to happen]
Compile to be successful once sensorless homing is supported with 2208s.

**Actual behavior:** [What actually happens]
Compile error due to sensorless homing only being supported on the 2130s.

#### Additional Information
Keep up the great work!

If I am mistaken about the 2208 support for StallGuard and there is no future for sensorless homing with 2208's, I will place an order for 2130s. :)

teemuatlut, 2018-05-07:
The TMC2208 doesn't support stallGuard. ([link](https://www.trinamic.com/products/integrated-circuits/))

comps, 2018-05-07:
Before ordering the 2130, note that the 2208 have a distinct advantage over 2130 - they support stealthChop2 instead of the "1" on 2130.

From what I was able to gather, the second version makes the motors/drivers run much cooler due to better current control algorithms, so if you can live with real endstops and run stealthChop, the 2208 might be better for you.

viper93458, 2018-05-07:
Thanks for the follow up. Back to mechanical endstops then. :)

closed 2018-05-07 by viper93458