MarlinFirmware/Marlin #10641: I just tried the Firmware Retract feature on my Anet A8 using Marlin.  While using it, i noticed an excess of extrusion after the G11.  I then tested the extruder and found the gearing was now incorrect.  Requesting 10mm of extrusion resulted in 5mm.

bailes17, 2018-05-07:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Lyr3x, 2018-05-07:
what a hell of issues description is that? 👎

bailes17, 2018-05-07:
Sorry but I am new to reporting issues

Stephen Bailey  |  Senior Staff Controls Engineer  |  Cadence, Inc.  |  Tel. 540-245-1011 x4411 | bailey@cadenceinc.com<mailto:bailey@cadenceinc.com> |
[cid:image001.png@01D3E5CF.F17F5040]<http://www.cadenceinc.com/> [cid:image002.png@01D3E5CF.F17F5040] <https://www.facebook.com/CadenceInfo?ref=hl&utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=Facebook>  [cid:image003.png@01D3E5CF.F17F5040] <https://www.linkedin.com/company/cadence-inc..?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=LinkedIn>  [cid:image004.png@01D3E5CF.F17F5040] <https://plus.google.com/b/102307290118200092994/+Cadenceincinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=GooglePlus>  [cid:image005.png@01D3E5CF.F17F5040] <https://www.youtube.com/user/cadenceinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=YouTube>
Confidentiality Notice: This information is confidential and intended for the recipient of this message only. If you have received this message in error please delete it and notify Cadence, Inc. at the address above.

Lyr3x, 2018-05-07:
There is a guideline, how to open and describe an issue or an feature request or whatever else

boelle, 2018-05-07:
after you have made title shorter and filled out the first post

connect with something like pronterface and send M502 followed by M500, this makes sure that settings in EEProm does not override those in firmware

also remove retract settings in slicer

now power cycle and try to extrude 50mm and check that 50mm is extruded

bailes17, 2018-05-07:
> why i say you have not filled out the first post is because it says this:

I could not figure out how to attach a zip file.
Also I need to get the .h files from my system at home to send the which means I can’t do it until tomorrow

Stephen Bailey  |  Senior Staff Controls Engineer  |  Cadence, Inc.  |  Tel. 540-245-1011 x4411 | bailey@cadenceinc.com<mailto:bailey@cadenceinc.com> |
[cid:image001.png@01D3E5E1.7B0A20E0]<http://www.cadenceinc.com/> [cid:image002.png@01D3E5E1.7B0A20E0] <https://www.facebook.com/CadenceInfo?ref=hl&utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=Facebook>  [cid:image003.png@01D3E5E1.7B0A20E0] <https://www.linkedin.com/company/cadence-inc..?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=LinkedIn>  [cid:image004.png@01D3E5E1.7B0A20E0] <https://plus.google.com/b/102307290118200092994/+Cadenceincinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=GooglePlus>  [cid:image005.png@01D3E5E1.7B0A20E0] <https://www.youtube.com/user/cadenceinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=YouTube>
Confidentiality Notice: This information is confidential and intended for the recipient of this message only. If you have received this message in error please delete it and notify Cadence, Inc. at the address above.

[First Step]
[Second Step]

you need to fill out that instead of the long title

boelle, 2018-05-07:
but you are still able to change the title

boelle, 2018-05-07:
also please edit your posts and remove the long blocks of not needed text

"Stephen Bailey  |  Senior Staff Controls Engineer  |  Cadence, Inc.  |  Tel. 540-245-1011 x4411 | bailey@cadenceinc.com<mailto:bailey@cadenceinc.com> |
[cid:image001.png@01D3E5E1.7B0A20E0]<http://www.cadenceinc.com/> [cid:image002.png@01D3E5E1.7B0A20E0] <https://www.facebook.com/CadenceInfo?ref=hl&utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=Facebook>  [cid:image003.png@01D3E5E1.7B0A20E0] <https://www.linkedin.com/company/cadence-inc..?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=LinkedIn>  [cid:image004.png@01D3E5E1.7B0A20E0] <https://plus.google.com/b/102307290118200092994/+Cadenceincinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=GooglePlus>  [cid:image005.png@01D3E5E1.7B0A20E0] <https://www.youtube.com/user/cadenceinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=YouTube>
Confidentiality Notice: This information is confidential and intended for the recipient of this message only. If you have received this message in error please delete it and notify Cadence, Inc. at the address above.
From: Bo Herrmannsen [mailto:notifications@github.com]
Sent: Monday, May 07, 2018 8:56 AM
To: MarlinFirmware/Marlin <Marlin@noreply.github.com>
Cc: Stephen Bailey <bailey@cadenceinc.com>; Author <author@noreply.github.com>
Subject: Re: [MarlinFirmware/Marlin] I just tried the Firmware Retract feature on my Anet A8 using Marlin 1.1.8. While using it, i noticed an excess of extrusion after the G11. I then tested the extruder and found the gearing was now incorrect. Requesting 10mm .."

of course to remove those blocks you need to be at a pc and not use email to reply

boelle, 2018-05-07:
you attach a zip by edit the first post and at the bottom of edit window you will see

[image]

boelle, 2018-05-07:
@Lyr3x in the spirit  of keeping post short maybe delete yours now that the title and first post has been corrected?

boelle, 2018-05-07:
@bailes17 do you still have problems after trying M502 & M500 ?

bailes17, 2018-05-07:
I can’t test it now.  I will test it tonight.
I will delete the post
thanks

Stephen Bailey  |  Senior Staff Controls Engineer  |  Cadence, Inc.  |  Tel. 540-245-1011 x4411 | bailey@cadenceinc.com<mailto:bailey@cadenceinc.com> |
[cid:image001.png@01D3E5EF.146B5800]<http://www.cadenceinc.com/> [cid:image002.png@01D3E5EF.146B5800] <https://www.facebook.com/CadenceInfo?ref=hl&utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=Facebook>  [cid:image003.png@01D3E5EF.146B5800] <https://www.linkedin.com/company/cadence-inc..?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=LinkedIn>  [cid:image004.png@01D3E5EF.146B5800] <https://plus.google.com/b/102307290118200092994/+Cadenceincinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=GooglePlus>  [cid:image005.png@01D3E5EF.146B5800] <https://www.youtube.com/user/cadenceinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=YouTube>
Confidentiality Notice: This information is confidential and intended for the recipient of this message only. If you have received this message in error please delete it and notify Cadence, Inc. at the address above.

bailes17, 2018-05-07:
How to Delete my post?

Stephen Bailey  |  Senior Staff Controls Engineer  |  Cadence, Inc.  |  Tel. 540-245-1011 x4411 | bailey@cadenceinc.com<mailto:bailey@cadenceinc.com> |
[cid:image001.png@01D3E5F0.21C55A90]<http://www.cadenceinc.com/> [cid:image002.png@01D3E5F0.21C55A90] <https://www.facebook.com/CadenceInfo?ref=hl&utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=Facebook>  [cid:image003.png@01D3E5F0.21C55A90] <https://www.linkedin.com/company/cadence-inc..?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=LinkedIn>  [cid:image004.png@01D3E5F0.21C55A90] <https://plus.google.com/b/102307290118200092994/+Cadenceincinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=GooglePlus>  [cid:image005.png@01D3E5F0.21C55A90] <https://www.youtube.com/user/cadenceinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=YouTube>
Confidentiality Notice: This information is confidential and intended for the recipient of this message only. If you have received this message in error please delete it and notify Cadence, Inc. at the address above.

bailes17, 2018-05-07:
Leave a comment

boelle, 2018-05-07:
small x

[image]

boelle, 2018-05-07:
why did you close?

bailes17, 2018-05-07:
I am trying to delete

Stephen Bailey  |  Senior Staff Controls Engineer  |  Cadence, Inc.  |  Tel. 540-245-1011 x4411 | bailey@cadenceinc.com<mailto:bailey@cadenceinc.com> |
[cid:image001.png@01D3E5F1.53C5D730]<http://www.cadenceinc.com/> [cid:image002.png@01D3E5F1.53C5D730] <https://www.facebook.com/CadenceInfo?ref=hl&utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=Facebook>  [cid:image003.png@01D3E5F1.53C5D730] <https://www.linkedin.com/company/cadence-inc..?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=LinkedIn>  [cid:image004.png@01D3E5F1.53C5D730] <https://plus.google.com/b/102307290118200092994/+Cadenceincinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=GooglePlus>  [cid:image005.png@01D3E5F1.53C5D730] <https://www.youtube.com/user/cadenceinfo?utm_source=Email%20Signature&utm_medium=Banner&utm_campaign=YouTube>
Confidentiality Notice: This information is confidential and intended for the recipient of this message only. If you have received this message in error please delete it and notify Cadence, Inc. at the address above.

read this page too: https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute

bailes17, 2018-05-08:
I Tried the M502 & M500 to no avail.  First I examined the generated code from Cura and found the reason for the over extrusion to be with the program.  
I generated the program for the same part first with gcode flavor set as "Reprap (Marlin/Sprinter)" next with machine gcode flavor set as "UltiGcode" to generate G10 and G11 for retracts.  as you can see the extruder commands generate a different extruder volume for move between the two gcode flavors.

Generated with "Reprap (Marlin/Sprinter)"
M107
G0 F9000 X98.280 Y98.280 Z0.200
;TYPE:SKIRT
G1 F1200 X121.720 Y98.280 E0.93554
G1 X121.720 Y104.720 E1.19258
G1 X104.721 Y104.720 E1.87104

Generated with "UltiGcode"
M107
G0 F9000 X98.280 Y98.280 Z0.200
;TYPE:SKIRT
G1 F1200 X121.720 Y98.280 E2.25024
G1 X121.720 Y104.720 E2.86848
G1 X104.721 Y104.720 E4.50038
I don't understand why this would be the case but I am looking at Cura for the over extrusion issue I originally reported.

boelle, 2018-05-08:
when set to Reprap (Marlin/Sprinter) is it also set to use firmware retract?

post some screen shots, you do that by pressing alt gr + prt scr keys

then here you simply press ctrl+v

bailes17, 2018-05-08:
No
When set to Reprap (Marlin/Sprinter) it uses slicer driven retract

boelle, 2018-05-08:
have you tried to set it to firmware retract when set to Reprap (Marlin/Sprinter)

bailes17, 2018-05-08:
No
	At that point Cura does not generate the M10 and M11, instead in generates E moves.  The firmware has warnings in it not to use automatic firmware retract which would attempt to replace slicer retracts with firmware retracts.  I am going for the simplest approach I can.  The reason for using the Firmware Retract feature is to gain the recovery additional distance because my parts show evidence that the recovery area is thin or nonexistent.  That feature is not available on my version of Cura.  I can not run the later version due to the age of my computer, but I have ordered a newer computer which should soon allow me to use the latest version of Curan and hopefully solve these issues.
Thanks

boelle, 2018-05-08:
or try slic3r PE https://www.prusa3d.com/slic3r-prusa-edition/

bailes17, 2018-05-08:
Thank You.  I downloaded slic3r and I can see where you can set it up for firmware retraction. 
I will try it out tonight.

thinkyhead, 2018-05-08:
Be sure to test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. We're always fixing things.

boelle, 2018-05-09:
you could try and dive more deep in settings in cura, it sounds very strange that when set to Reprap (Marlin/Sprinter) you can't select firmware retract

the only reason why i dont use cura is that they removed manual select of serial port

bailes17, 2018-05-09:
I could but I really like slice3r you turned me on to, it gave much better results and has better features.
it also  ran on my old computer just fine.
Thank You

boelle, 2018-05-09:
Your Welcome :-)