MarlinFirmware/Marlin #10642: [IDEA] Make UBL detect abnormal / false readings

boelle, 2018-05-07:
so i have been using UBL for a bit now and for the most part its great, have never seen better probing

But i have experienced spots where it triggers 5mm above everything else

That made me look at the G26 pattern and think this idea:

if there is a single spot that change more than Xmm move the 8 probe poins arround it closer, say half way to the problematic spot. if reading are still ok and do not change much move half way closer again.

of course at some point you are so close to the problematic spot that you can make an educated guess that the problematic spot should be

my thought behind this is if it can be done more automatic then the user has less fiddlework to do

Roxy-3D, 2018-05-07:
Or...    Another approach might be to add some more logic to the UBL Sanity Check.   There are a couple of possibilities.    For starters, checking that the mean height of the bed's mesh points is between -1.0mm and +1.0mm would be valuable.    And then checking that no mesh point is further than 1.0mm from the mean probably makes sense.

It would not be that hard to print out a caution message if the mesh point values are less than trust worthy.   But there is already so much clutter printed out, I'm not sure people will see the messages.

gloomyandy, 2018-05-07:
Identifying suspicious probe values is a good idea. However I'm not sure that automatically correcting them would be a good thing. It is probably better to understand what is causing them (are they repeatable etc.), because whatever is causing the "very large error" may also be causing slightly less large errors in other readings and those slightly less large errors can still be enough to crash the nozzle into the bed.

boelle, 2018-05-07:
automatic correcting them without asking the user i tend to agree

in my case i use an difirential ir probe that sometimes have a spot where it triggers 5 mm before every other mesh point

but if i start the probing at say X50Y55 it will not trigger "false" when it comes to the trouble point in the middle of the bed where the maker Orballo Printing "wisely" have put their name/logo in white blank silkscreen.. 

so if i start the probing from homed (G28) i get a problem in the center
if i start from X50Y55 i dont get a problem in the center... very odd i know

Roxy-3D, 2018-05-07:
I agree 'correcting' values without the user saying to do that is inherently dangerous.   Mostly...   the thinking behind a more detailed Sanity Check is to catch obvious configuration errors.   But if an inductive probe mis-fires and generates a crazy value, that would be found too and the user would be alerted.

teemuatlut, 2018-05-07:
@Roxy-3D Was there a command to re-probe just one point in the UBL mesh?

Roxy-3D, 2018-05-07:
There probably should be an easy way to do that.   The reason that is missing is the 'philosophy' was to probe and fill out the entire mesh.    You can do it indirectly by invalidating a mesh point, and then reprobing the unprobed locations of the mesh (with G29 P1 C).    

Also...  part of the explanation for not being able to probe a single point directly is if a single mesh point is bad...  it is sort of obvious what it should be by looking at the values around it.    I guess I never ended up in the situation where I wanted to probe a single location.   (With the exception of manually probing a few locations in the unreachable area of the bed.)

Roxy-3D, 2018-05-07:
Yeah...   For now...  You can look at all the probed points, and invalidate 1 or 2 or 3 of them.   And then do a G29 P1 C to fill out the mesh.  But I do agree, being able to probe a single point makes sense.   The hard part is communicating to the user when that should be used.

boelle, 2018-05-07:
would an alert on LCD fit the bill?

boelle, 2018-05-07:
ie an alert saying:

 "abnormal" probes found. Reprobe ? 

and then user can click yes or no

boelle, 2018-06-03:
[image]

a quick hand drawn illustration.

the read circle is the probe that differs to much and are marked abnormal

i could picture the probing be done like this:

- probing is done as normal
- probing should not be aborted if it fails to probe a single point
- at the end of probing  it should say: Abnormal probes found, reprobe ?
- if user say yes it shoould now probe at the green circles instead and then from those probes determine where the abnormal probes are or should be

Roxy-3D, 2018-06-03:
It is best to home in the center of the bed, so that is a bad place to have the probe be inaccurate.
How consistent is its reading there (with M48)?    You have an inductive probe, right?
Is there extra metal there or something causing the reading to be inaccurate?

boelle, 2018-06-04:
The red circle is not in the center of the bed, it could be ANY abnormal probe point on the bed

0.01 with M48, i have both inductive an a differential ir probe (it uses 2 different ir emitters to be immune to sunligt etc)

boelle, 2018-06-04:
what could cause these readings to go off could be if a probe point happens to be on top of one of the 6 screws used to hold the bed, that will cause an inductive probe to trigger a bit different.

in case of the IR probe the screw can also reflect the light differently should it happen that the probe is done on top of the screw.

also the white silkscreen could throw off the readings in case of the ir as the silkscreen is not  mat but rather shiney.

I dont have piezo sensors like @teemuatlut  have put i guess those are even more sensitive than what i have, so the feature is not just for my own use

boelle, 2018-06-04:
for the fun of it i have tried to do a probe with the inductive probe, that is not perfect either, if triggers differently when it probes on top of one of the screws, and it triggers slightly higher up if it hits one of the 9 probe points dead center. So 3 trigger heights depending if a probe point is on top of a screw or probe point or if anywkere else

Roxy-3D, 2018-06-04:
If the machine homes the same every time, I might not worry about the Z-Probe.   The reason being, with the mesh based bed leveling systems, you can save the mesh, and edit out the bad readings (that were used to generate the mesh).

If the probe is returning a .01mm standard deviation with M48, it is hard to use its values to detect a bad reading.   The only other thing that could be done is to use surrounding values to determine a location's value is wrong.   But that still doesn't help us get an accurate value for that location.

boelle, 2018-06-04:
> . But that still doesn't help us get an accurate value for that location.

Nope and we will never be able to, in case of the MK42 bed an inductive probe will have 6 screw heads that will trigger a slightly different reading. only fix is to use plastic screws and i dont think they will hold up very well. 

This is why i suggested the "narrow it down way", it will not be perfect put if you can get a "good" reading say only 5mm away from the bad one i think we are better off than with manual work

Roxy-3D, 2018-06-04:
I think a better approach would be to take what ever the reading is....   And use it as a starting point.
G26 is setup to be used in an iterative manner.   You will see the mesh point value get better and better as you edit and run a subsequent G26 Mesh Validation Pattern.

boelle, 2018-06-04:
why not try and make the machine figure it out for the user? I'm not thinking for my own use here

for a new user the experience might push them away if they have to spend maybe an hour extra with multiple test prints

and if the machine only needs to probe say 2-3 mm away from a bad probe and can do 4 good probes arround a bad one i think the interpolated value will be good enough

but of course it will require some coding and i can see that for people with setups where there are only 4 corner screws and a silicone heater under an alu plate it might never be needed

Roxy-3D, 2018-06-18:
If you know a mesh point is bad...   You probably want to invalidate it and re-probe it.    Or...   You can just edit it to get it closer to correct.     Ultimately, it is the G26 mesh validation pattern that should be trusted.    You need the mesh values close enough you can do a G26.    But G26 will show you what is really happening.

boelle, 2018-06-18:
reprobe is not an option as it would just give me the same outcome

manual edit kind of defeats the purpose of an automated system

what happens if I invalidate and use G29 P3 T ? will it then calcualte an value or what happens?

gloomyandy, 2018-06-18:
Give it a try and see if you like the results (I honestly don't know as I only use G29 P3 to fill in missing values after a probe). You might like to try the G29 P3.1 options...
```
 // P3.1  use least squares fit to fill missing mesh values
                  // P3.10 zero weighting for distance, all grid points equal, best fit tilted plane
                  // P3.11 10X weighting for nearest grid points versus farthest grid points
                  // P3.12 100X distance weighting
                  // P3.13 1000X distance weighting, approaches simple average of nearest points
```
These do not seem to be documented, but I prefer the results over the G29 P3 smart fill operation.

boelle, 2018-06-18:
problem is that most of these points are on top of a screw which triggers different since its steel and not copper. and then the bed has 9 points made of copper also which tiggers different than the screws or bed since they have a largers area.

so G29 P3.13 would calculate an value from the nears points arround the missing one?

gloomyandy, 2018-06-18:
G29 P3.1x basically uses a weighted least squares fit with a higher weighting being given to those points nearest the missing one. The x lets you control the weighting making the nearest points more or less important. I've had good results with both 3.12 and 3.13.

boelle, 2018-06-18:
👍 that paired with the octoprint plugin will make it a walk in the park. but had i known before hand i would have gotten a bl touch, which i for sure will get for my 500x500x500mm printer

boelle, 2018-06-18:
closing this one as i found a way arround my "problem" in a more automated way

closed 2018-06-18 by boelle

Roxy-3D, 2018-06-18:
> but had i known before hand i would have gotten a bl touch, 

Get a 'Real' one.   Avoid the clone BL-Touch probes.   Especially if your 500mm x 500mm machine is 'nice'.   Nozzle crashes do a lot of damage.

dough29, 2018-06-18:
Genuine is far the best !
I tested both :
- clone : min accuracy = 0,020mm / MAX = 0,067mm / average = 0,043mm
- BLTouch : min = 0,007mm / MAX = 0,020mm / average = 0,011mm
Both sensors have been tested on 45 measures.

boelle, 2018-06-18:
> Genuine is far the best

its the only right thing to do, i avoid supporting ripoffs whenever i can even if it cost me more. Better wait and get the good stuff

boelle, 2018-06-18:
thou in case of my bed i cant get the mk42 from prusa as he requires that i have his printer before he sells the heated bed to me

in that case i got it from orballoprinting.com so its a clone, but its a flat one and not from china.

boelle, 2018-06-28:
got time to test that octoprint plugin

[image]

on the right you can see where it hits one of the 9 probe points and hence create a spike, it looks worse than it is but it shows what i mean

nice thing with the plugin i can now just point at that spike and get the cord's for it and then with gcode move to that spot and invalidate it and use G29 P3.13 to have the firmware figure out a more "reasonble" number

it might not be perfect but its a hell lot faster than farting arround with G26