MarlinFirmware/Marlin #10643: Strange UBL behavior

smoki3, 2018-05-07:
I updated my printer at the weekend to the actual 2.x branch from a Februar (also 2.x) build. 

Then I tried to redo my bed leveling. I started to regenerate the UBL mesh with a G29 P1.

But my printer is only probing **4 points**. On the old config build it was probing the whole bed. 

So is the behavior changed? Or is there a reason why its only probing 4 points?

Config file: https://pastebin.com/bssuC37e

ManuelMcLure, 2018-05-07:
The probing logic has changed recently - it now takes into account MIN_PROBE_EDGE (10 in your configuration). MIN_PROBE_EDGE is a configuration setting that ensures that the probe doesn't check too close to the edge of the bed (inductive and capacitive sensors tend to be quite inaccurate if too close to the edge of the bed). Since your MESH_INSET is smaller than MIN_PROBE_EDGE it can't probe the points closest to the edge of the bed and you only get the four middle points.

If you have a sensor like the BLTouch you can reduce the value of MIN_PROBE_EDGE since those do not have any problems probing the very edge of the bed. If you have a capacitive or inductive sensor it's not a good idea to reduce MIN_PROBE_EDGE - instead either increase MESH_INSET to 10 (this will reduce your printing area) or increase the number of points in your mesh. The latter is probably best - you still won't be able to probe the very edge, but you can use G29 P3 to fill in the unprobed spots and fine tune them with G26 and G29 P4. I use a 10x10 grid with MESH_INSET of 2 and MIN_PROBE_EDGE of 15 with this method.

smoki3, 2018-05-07:
So now I am confused. You said if my mesh_inset is smaller than the min_probe_edge than I only get the four points. 

But you are using Mesh_inset 2 and min_probe_edge 15. So you also get only the four point? Because you inset is the smaller value?

Roxy-3D, 2018-05-07:
Can you post a G29 T W      That will show us what it probed and where it thinks the mesh points are located.

smoki3, 2018-05-08:
`Send: N13 G29 T W*35
Recv: echo:Vorher Z homen
Recv: Unified Bed Leveling System v1.01 active.
Recv: Mesh 0 Loaded.
Recv: UBL object count: 1
Recv: planner.z_fade_height : 10.0000
Recv: # of samples: 25
Recv: Mean Mesh Height: -0.029700
Recv: Standard Deviation: 0.021252
Recv: zprobe_zoffset: -1.0900000
Recv: MESH_MIN_X  (((((((210) / 2) - (210) / 2) + 8) > (-28)) ? ((((210) / 2) - (210) / 2) + 8) : (-28)))=8
Recv: MESH_MIN_Y  (((((((200) / 2) - (200) / 2) + 8) > (0)) ? ((((200) / 2) - (200) / 2) + 8) : (0)))=8
Recv: MESH_MAX_X  (((((((210) / 2) + (210) / 2) - (8)) < (230)) ? ((((210) / 2) + (210) / 2) - (8)) : (230)))=202
Recv: MESH_MAX_Y  (((((((200) / 2) + (200) / 2) - (8)) < (200)) ? ((((200) / 2) + (200) / 2) - (8)) : (200)))=192
Recv: GRID_MAX_POINTS_X  5
Recv: GRID_MAX_POINTS_Y  5
Recv: MESH_X_DIST  48.50
Recv: MESH_Y_DIST  46.00
Recv: X-Axis Mesh Points at: 8.000  56.500  105.000  153.500  202.000
Recv: Y-Axis Mesh Points at: 8.000  54.000  100.000  146.000  192.000
Recv: 
Recv: Unified Bed Leveling sanity checks passed.
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     (  8,192)                      (202,192)
Recv:         0       1       2       3       4
Recv:  4 |[-0.055] -0.055  -0.009  -0.036  -0.036
Recv:    |
Recv:  3 | -0.055  -0.055  -0.014  -0.042  -0.042
Recv:    |
Recv:  2 | -0.051  -0.051  -0.018  -0.048  -0.048
Recv:    |
Recv:  1 | -0.037  -0.037  -0.011  -0.018  -0.018
Recv:    |
Recv:  0 | -0.022  -0.022  -0.004  +0.012  +0.027
Recv:         0       1       2       3       4
Recv:     (  8,  8)                      (202,  8)
Recv: `

Roxy-3D, 2018-05-08:
This is a 5 x 5 mesh?    And all points are present?

Roxy-3D, 2018-05-08:
Recv: GRID_MAX_POINTS_X  5
Recv: GRID_MAX_POINTS_Y  5
Recv: MESH_X_DIST  48.50
Recv: MESH_Y_DIST  46.00
Recv: X-Axis Mesh Points at: 8.000  56.500  105.000  153.500  202.000
Recv: Y-Axis Mesh Points at: 8.000  54.000  100.000  146.000  192.000

smoki3, 2018-05-08:
But shouldn't it probe 5x5 (25) points?

smoki3, 2018-05-08:
On my previous firmware it was probing many many point on the whole bed. no only 4 points

thinkyhead, 2018-05-13:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed.
- Copy the log output into a **.TXT file** and attach it to your next reply.

smoki3, 2018-05-17:
If comment out  `min_probe_edge` it works as expected

closed 2018-05-17 by smoki3

thinkyhead, 2018-05-22:
Interesting! Same if you set it to 0?