MarlinFirmware/Marlin #10646: [bugfix-2.0.x] Compile error when #define TEMP_SENSOR_0 -3

boelle, 2018-05-07:
So i got the MAX31855 board wired in for the hotend

changed thermistor setting to -3 and compile failed:

```

In file included from C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\thermistor/../../inc/../HAL/HAL_AVR/HAL.h:47:0,

                 from C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\thermistor/../../inc/MarlinConfig.h:28,

                 from C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\thermistor/thermistors.h:26,

                 from C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\temperature.h:30,

                 from C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\temperature.cpp:27:

pins_RAMPS.h:209:33: error: pasting "/* Do not use pin 49 as this is tied to the switch inside the SD card socket to detect if there is an SD card present*/" and "_DDR" does not give a valid preprocessing token

   #define MAX6675_SS       66   // Do not use pin 49 as this is tied to the switch inside the SD card socket to detect if there is an SD card present

                                 ^

C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\thermistor/../../inc/../HAL/HAL_AVR/fastio_AVR.h:84:42: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)

                                          ^

C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\thermistor/../../inc/../HAL/HAL_AVR/fastio_AVR.h:102:35: note: in expansion of macro 'SET_OUTPUT'

 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)

                                   ^

C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\temperature.cpp:1189:5: note: in expansion of macro 'OUT_WRITE'

     OUT_WRITE(MAX6675_SS, HIGH);

     ^

C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\temperature.cpp:1189:15: note: in expansion of macro 'MAX6675_SS'

     OUT_WRITE(MAX6675_SS, HIGH);

               ^

pins_RAMPS.h:209:33: error: pasting "/* Do not use pin 49 as this is tied to the switch inside the SD card socket to detect if there is an SD card present*/" and "_PIN" does not give a valid preprocessing token

   #define MAX6675_SS       66   // Do not use pin 49 as this is tied to the switch inside the SD card socket to detect if there is an SD card present

                                 ^

C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\thermistor/../../inc/../HAL/HAL_AVR/fastio_AVR.h:84:61: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)

                                                             ^

C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\thermistor/../../inc/../HAL/HAL_AVR/fastio_AVR.h:102:35: note: in expansion of macro 'SET_OUTPUT'

 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)

                                   ^

C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\temperature.cpp:1189:5: note: in expansion of macro 'OUT_WRITE'

     OUT_WRITE(MAX6675_SS, HIGH);

     ^

[... 758 lines cut ...]
     ^

C:\Users\boher\AppData\Local\Temp\arduino_build_461408\sketch\src\module\temperature.cpp:1642:11: note: in expansion of macro 'MAX6675_SS'

     WRITE(MAX6675_SS, 1); // disable TT_MAX6675

           ^

Using library U8glib at version 1.19.1 in folder: C:\Users\boher\Documents\Arduino\libraries\U8glib 
exit status 1
Exception in thread "Thread-380" java.lang.ArrayIndexOutOfBoundsException: -1
	at java.util.ArrayList.elementData(ArrayList.java:422)
	at java.util.ArrayList.get(ArrayList.java:435)
	at processing.app.Editor.getCurrentTab(Editor.java:1606)
	at processing.app.Editor.updateUndoRedoState(Editor.java:1559)
	at processing.app.Editor.selectTab(Editor.java:1632)
	at processing.app.Editor.statusError(Editor.java:2756)
	at processing.app.Editor$BuildHandler.run(Editor.java:1827)
	at java.lang.Thread.run(Thread.java:748)

```

are there other places i need to change things to make it work?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1981692/Configuration.zip)

boelle, 2018-05-07:
in the zip is also the pin file for ramps

the only thing in there that relates to the error is:

```
// SPI for Max6675 or Max31855 Thermocouple
#if DISABLED(SDSUPPORT)
  #define MAX6675_SS       66   // Do not use pin 53 if there is even the remote possibility of using Display/SD card
#else
  #define MAX6675_SS       66   // Do not use pin 49 as this is tied to the switch inside the SD card socket to detect if there is an SD card present
#endif
```

but that it just does not makes sense for me

p3p, 2018-05-07:
It's the Arduino IDE bug  which means the compile fails because of comments on the same line as defines, remove the comment

> but that it just does not makes sense for me

It makes no sense to anyone why Arduino (or maybe avr-gcc) seems to have developed this oddity in its preprocessor, you can tell its this problem whenever you see `error: pasting`

Grogyan, 2018-05-07:
I have already raised this as an issue #8010

p3p, 2018-05-07:
@Grogyan Your issue is unrelated, caused by a missing definition in Re-ARM pins file

Grogyan, 2018-05-07:
I had tried a long time ago with a standard MEGA and RAMPS, to the same effect.
So because I can run with the MEGA with under the 1.x moniker, it hasn't been a huge priority for me.

So am guessing that ReArm and MEGA under 2.x is going to cause the same for the MAX31855.

boelle, 2018-05-08:
So can this be called a bug ?

boelle, 2018-05-08:
ahh... yet another one, remember i had a similar issue with another pin

boelle, 2018-05-08:
compiled and flashed, but now it goes straight to min temp error or max temp error if max chip is disconnected, so at least i did connect it to the right port

will try and disable it and see what temp it gets

boelle, 2018-05-08:
still getting max temp error

[image]

that is there i have connected the max chip, just checked it again

boelle, 2018-05-08:
the error

[image]

boelle, 2018-05-08:
that was with the min temp set to crazy -50000 and max set to crazy 52750

with min temp set to 5 and max set to 275 i get this on serial

boelle, 2018-05-08:
when i used a normal thermistor i did not get any errors even if i disconnected it, the temp was just 0 in that case

and my head cant figure how to disable this check, would like to see what the temp reads out to

boelle, 2018-05-08:
so back to normal settings i started to look at the "hidden" files (those that do now show up in the IDE

\src\module\temperature.cpp

i commented out line 803 & 804

   // if (current_temperature[0] > min(HEATER_0_MAXTEMP, MAX6675_TMAX - 1.0)) max_temp_error(0);
   // if (current_temperature[0] < max(HEATER_0_MINTEMP, MAX6675_TMIN + .01)) min_temp_error(0);

now id there is no max chip connected i reports 800 degree

if i connected a max chip it alternates between 0 and 800

as i have prepared a plug for the bed with CS being D65 i tried both plugs and the outcome was the alternating between 0 and 800 on both

will try and recheck supply voltage

boelle, 2018-05-08:
hmm... just connected with octoprint to check serial

Recv: Error:Temp measurement error! MAX31855 Open Circuit

boelle, 2018-05-08:
will check for that and voltage

boelle, 2018-05-08:
i checked for that its now more stable, all i did was tighten the screw terminals on the tiny max breakout board

i now get:

 ```
Recv:  T:1800.00 /0.00 B:22.50 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.50 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.47 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.50 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.50 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.50 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.47 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.53 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.47 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.47 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: Error:Temp measurement error! MAX31855 Open Circuit
Recv: Error:Temp measurement error! MAX31855 Open Circuit
Recv: Error:Temp measurement error! MAX31855 Open Circuit
Recv:  T:0.00 /0.00 B:22.47 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.44 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.47 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.50 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.50 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.47 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.53 /0.00 @:0 B@:0
Recv: Error:Temp measurement error! MAX31855 Open Circuit
Recv: Error:Temp measurement error! MAX31855 Open Circuit
Recv: Error:Temp measurement error! MAX31855 Open Circuit
Recv: Error:Temp measurement error! MAX31855 Open Circuit
Recv: Error:Temp measurement error! MAX31855 Open Circuit
Recv:  T:0.00 /0.00 B:22.50 /0.00 @:0 B@:0
Recv:  T:0.00 /0.00 B:22.50 /0.00 @:0 B@:0
```

boelle, 2018-05-08:
could it be that my max chip is not the MAX31855 but the MAX31855K ?

boelle, 2018-05-08:
i rechecked everything again, no obvisous errors found 

i checkeced every single wire from the ramps board to the max breakout and one of the dupont crimps would not go on all the way so will crimp on a new connector later

but funny enough without the CS wire attached i get -39 degree's

boelle, 2018-05-08:
also when i read forums etc there is a lot of confusion on how to connect

some say that miso and mosi needs to cross

its a board that looks like this

[image]

boelle, 2018-05-08:
This is how i understand it

SO is ouput at 3.3v which is high enough that the Mega can read it as high or low
CS is chip select from the Mega and i need to get that down to 3.3V 
SCK also needs to be brung down to 3.3V

since its only one way i would assume resistor can do it at a pinch, a level shifter is better but it takes longer to get here

Bob-the-Kuhn, 2018-05-08:
The MAX31855 is 5V tolerant so no need for a level shifter.

SO goes to MISO (Master In, Slave Out).

boelle, 2018-05-08:
is the MAX31855K also tolerant?

boelle, 2018-05-08:
https://www.maximintegrated.com/en/support/faqs/max31855-faq.html

> Can I drive the MAX31855 SPI pins with a 5V supply?
> No, the MAX31855's SPI interface must not have voltage higher than 4.0V applied. High voltage andhigh current (through a voltage divider) will induce self-heating resulting in cold-junction andthermocouple temperature errors, as well as possible damage to the SPI interface pins.

boelle, 2018-05-08:
but at least i got connectins right the first time 

before i used voltage dividers it said 800 degree's constant, not it's anywhere from -90 to -10

boelle, 2018-05-08:
to rule out i damaged i solder on screw terminal and header on another breakout and use the other thermistor.

i ordered 4 of those breakout as they where cheap and i futureproofed my stock with a thermocouple cartridge from e3d and a normal thermocouple (so i could put one on bed when/if it becomes an option)

output on serial is now 

```
Recv:  T:-1506.00 /0.00 B:21.16 /0.00 @:0 B@:0
Recv:  T:-1422.00 /0.00 B:21.19 /0.00 @:0 B@:0
Recv:  T:-1452.00 /0.00 B:21.16 /0.00 @:0 B@:0
Recv:  T:-1512.00 /0.00 B:21.12 /0.00 @:0 B@:0
Recv:  T:-1536.00 /0.00 B:21.19 /0.00 @:0 B@:0
Recv:  T:-1440.00 /0.00 B:21.09 /0.00 @:0 B@:0
Recv:  T:-1460.00 /0.00 B:21.16 /0.00 @:0 B@:0
Recv:  T:-1424.00 /0.00 B:21.09 /0.00 @:0 B@:0
```

Bob-the-Kuhn, 2018-05-09:
You are correct, the MAX31855 is not 5V tolerant.

I was originally looking at a breakout board but further reading said it had logic shifters on the board.

boelle, 2018-05-09:
:-) now i just have to figure why it reports those crazy negative numbers back

TadyTheFish, 2018-05-10:
Try swapping the thermocouple wires around?

boelle, 2018-05-10:
tried that
tried another thermocouple (have 2 in total of same type, K type)
tried another breakoutboard (have 4 )
tried 10nf cap on the thermocouple terminals

will try and hook up the breakout boards to an arduino uno and load a max31855 test sketch and see if it also reports "crazy" numbers there

TadyTheFish, 2018-05-10:
How long are your wires for the SPI com? I don't know the SPI is running in Marlin but it must be 1MHz or more.. So any long wires act like an antenna. But that would probably give you corrupt data...

boelle, 2018-05-10:
if i unplug the breakout board i get a constant -99 and if i plug in the breakout with no thermocouple the output on serial just says thermocouple disconntected

but give me an hour and i will see if its the same on the uno

TadyTheFish, 2018-05-10:
Thats what I was just about to say.. Load a test sketch..

boelle, 2018-05-10:
getting a solid 2047.75 so far

how often do marlin ask what temp the hotend is, i can asjust my sketch to do the same interval

TadyTheFish, 2018-05-10:
Isn't that wrong? What is your actual temp

boelle, 2018-05-10:
20'ish

boelle, 2018-05-10:
at what interval does marlin ask for temp? i guess a lot faster than every 3 sec

boelle, 2018-05-10:
anyways i have ordered a sparkfun branded breakout that will be here either tomorrow or monday. sadly no mail today as its Christ's ascension (i hate holidays in general)

boelle, 2018-05-10:
@thinkyhead is there anything we can do about the GCC errors? if not i'm tempted to close this one as my readings on an uno are the same as when connected to printer and i'm 90% sure that its a bad or counterfeit chip or just a bad designed breakout

thinkyhead, 2018-05-10:
>  is there anything we can do about the GCC errors? 

As already explained, those are due to a bug in GCC. As a workaround, you should remove end-of-line comments from any pins that cause errors for you. We are _not_ going to edit all the pins definitions throughout Marlin.

closed 2018-05-10 by thinkyhead

Grogyan, 2018-05-10:
As mentioned from the Facebook group, if there are caps on the thermocouple lines, and ensure that there is 3.3V going into the board.

If still not working, then it is likely that they have counterfeit devices on them.