MarlinFirmware/Marlin #10647: Feature: Add support for Bigtreetech/BIQU KFB 2.0 controller board

adamrblack, 2018-05-08:
### Description

The controller board I have is a Bigtreetech or BIQU "KFB 2.0"
It's basically a RAMPS 1.4 AIO using the MKS_GEN_L board standard (without any servo connections)

basically, all a pins_KFB_2.h would contain would be

```
/**
 *KFB 2.0 – Arduino Mega2560 with RAMPS v1.4 pin assignments
 */

#if HOTENDS > 2 || E_STEPPERS > 2
  #error "KFB 2.0 supports up to 2 hotends / E-steppers. Comment out this line to continue."
#endif

#define BOARD_NAME "KFB 2.0"

//
// Heaters / Fans
//
// Power outputs BEEF or BEFF
#define MOSFET_D_PIN 7

#include "pins_RAMPS.h"

```
and in boards.h
`
#define BOARD_KFB_2         "some number"    //Bigtreetech or BIQU KFB2.0
`

and in pins.h, in the "Ramps Derivatives" section

`
#elif MB(KFB_2)
  #include "pins_KFB_2.h"         // ATmega2560
`

Cheers,

Adam

skcks, 2018-06-24:
@adamrblack could you share your the config files of BIQU KFB 2.0 ？

adamrblack, 2018-06-24:
in Configuration.h i have 

```
// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_KFB_2
#endif
```

in pins_KFB_2.h (new file) i have

```
/**
 *KFB 2.0 – Arduino Mega2560 with RAMPS v1.4 pin assignments
 */

#if HOTENDS > 2 || E_STEPPERS > 2
  #error "KFB 2.0 supports up to 2 hotends / E-steppers. Comment out this line to continue."
#endif

#define BOARD_NAME "KFB 2.0"

//
// Heaters / Fans
//
// Power outputs BEEF or BEFF
#define MOSFET_D_PIN 7

#include "pins_RAMPS.h"
```

in boards.h I have added 

`#define BOARD_KFB_2 "369" //Bigtreetech or BIQU KFB2.0`

and in pins.h, in the "ramps 1.4 derivatives section i have added

`#elif MB(KFB_2) #include "pins_KFB_2.h" // ATmega2560`

Attached is a Zip with the changes I've made
[Marlin-KFB2.zip](https://github.com/MarlinFirmware/Marlin/files/2130549/Marlin-KFB2.zip)

adamrblack, 2018-06-24:
you can also just set your "MOTHERBOARD MKS_GEN_L" in Configuration.h, as its basically the same thing (but with no servo support)

skcks, 2018-06-24:
is that board supports 24v  power supply?

adamrblack, 2018-06-24:
Yes

closed 2018-08-03 by thinkyhead

tsukadaster, 2018-08-06:
@adamrblack Thanks a lot for the Marlin configuration! I successfully updated the firmware for my A8M which is dual extruter with KFB2.0. However, i found "Extruder Extruder 0 Extruder1" in the machine menu and none of these steppers can be manually controlled. Is there any possible reason for it? Thanks for help!

adamrblack, 2018-08-06:
You should be able to extruder just fine using the menus. Extruder's wont work untill a safe temp is reached (like 170ish) Make sure your right hand extruder is reversed for the A8M

tsukadaster, 2018-08-07:
Thanks a lot for the tips, it works fine! Thanks a lot!

adamrblack, 2018-11-25:
That's more of a configuration.h issue, invert the axes that are t moving the right direction. Better to ask on the Anet Facebook support page

maher01, 2018-12-02:
i need help , anyone have marlin firmware to anet A8M dual extruder, or someone can guide me what i need to do :?

gharryh, 2018-12-22:
So I updated the software to MARLIN 1.1.9 and the connected 12864 display say's printer ready.
But ther's no movement on any of the axis. I have connected only the power supply and the motors.
Or shuold I conect the thermisters also?

cassiosilvestre, 2018-12-22:
> So I updated the software to MARLIN 1.1.9 and the connected 12864 display say's printer ready.
> But ther's no movement on any of the axis. I have connected only the power supply and the motors.
> Or shuold I conect the thermisters also?

Hi, share your configuration.h with us this way makes it easier to solve the problem

nmong0415, 2019-01-12:
i also have the a8-m and am looking to update it and add the auto level sensor, i am working on a bracket for it as we speak

nmong0415, 2019-01-12:
any luck resolving the config file issue? or was it a physical issue?

EmersonPombo, 2019-06-01:
Alguém poderia me ajudar com minha Anet A8M KFB2.0 bord, gostaria do firmware original dela !
Alguém tem ?
Sou leigo nesta área !

richardpub4136, 2019-07-16:
The problems with my AnetA8-M on which the SD Card presents with the LED screen, displays "No SD Card" although I can open this card, see the folders and files, but I am unable to print them, so everything works with the USB cable connected to my PC.
The motherboard of the AnetA8-M is a KFB2.0. I'm trying to install Marlin with AVRDUDESS.exe, but I can not connect.
I would not want to burn my card with bad handling.
Anet.com sent me a tutorial to inject the original .hex file and an .hex test file. They tell me to choose Arduino as a map, but is it really the right choice. Visibly the COM port is present, but it does not recognize the card.
AVRDUDESS recognizes the presence of the port as soon as it is connected but I have no choice of KFB2.0
I receive this error massage:
 "avrdude.exe: stk500_getsync () attempt 10 of 10: not in sync: resp = 0x3d
avrdude.exe done. Thank you. "

Should I go through the Arduino IDE ??

Thanks for your advices.

adamrblack, 2019-07-16:
Use the Arduino ide. Download a release, or Clint the repo, configure for your printer and flash