MarlinFirmware/Marlin #10655: Dual endstop and auto bed leveling

Koby88, 2018-05-08:
Hi all first i want to apologize for my bad English because it is not my native language...
Now i want to know if someone can help me with my Z homing..
I have bltouch and i want to use additional two z axis endstops because i have two separate lead screw on z axis and i want to occasionally make them same level. I know how to do that by enabling dual endstop in marlin, what i dont know is how to setup that bltouch is used for probing (min endstop) and two others endstop for Z max so i can disable it throughg code, coz i dont need every time to go z max just to synchronize them.
I want if it is possible that my g code look like this:
G28 X0; home x axis
G28 Y0; home y axis
G28 Z0; home z axis to bltouch
G28 E0; for home z to max position so i can delete this command when im not using it...
I hope you can understand me and that you gonna help me if you can, thanks in adwance

thinkyhead, 2018-05-08:
If your Z endstops are set at the max positions, then `G28 Z` will always home to max. There's no option to switch to the probe for Z homing at this time.

thinkyhead, 2018-05-10:
> but i want to use my Z min like default G28 Z0, and to assign new letter to Z max endstops

No such option exists at this time. You may want to find a developer for hire who can implement this for you. It's unlikely to ever be added as a standard feature to the main Marlin project.

closed 2019-05-16 by thinkyhead