MarlinFirmware/Marlin #10658: Small corexy size bug ? or is it me, going crazy here :)

Spawn32, 2018-05-08:
Really need some help here.

I have a Tronxy X5s corexy with a mks sbase and bugfix 2.0.

My problem is that whatever i do mechanically i always end up with yellow side ( see link) being 0.2mm
bigger than the other side, the red side is spot on:

calib link: http://forums.reprap.org/read.php?397,530210

I have re tension'd the belts 10 times, all my pulleys are the same size, motors are the same brand and modell.
The gantry is a bit weak on the X5S so i printed a new strong one and was sure it would fix it, but again, 0.2mm wrong...

also tested with the original controller (a melzi with bugfix 2.0 ), same thing, 0.2mm

I have also taken the frame apart several times t be sure it is 100% square and it is :) also re enforced all corners..

Since there seems to be no other option, could this be a slight firmware bug ?
The really strange thing is that it dosen mater if i print a 20x20mm cube or a 40x40, it's always 0.2mm wrong..

anyone else seeing this ? or could some corexy user test the stl for me ? include here...

I know 0.2mm is not much, but i am going to print a planetary gearbox in fiber nylon and need 0.05mm accuracy :/

If not a bug, is there some way to tweek / force  this in config ? would be easy with a normal printer, but the corexy's are problematic with this kind a stuff :/

[testcube-40x40x10x5.zip](https://github.com/MarlinFirmware/Marlin/files/1985361/testcube-40x40x10x5.zip)

thinkyhead, 2018-05-09:
It may indicate a backlash issue. You can test this by doing a move in one direction, then trying a short (0.2mm) move in the other direction. If the axis doesn't move, then there's too much slack or slippage somewhere in the drive system.

If you print the cube in a diagonal orientation it will probably come out shorter from corner-to-corner, but may have better dimensions from edge-to-edge.

At this time there's no backlash compensation in Marlin, though it may be added later. Backlash compensation for CoreXY may present some extra challenge.

Spawn32, 2018-05-09:
Thank you for the input @thinkyhead 
have now checked for backlash on y +/- and x +/- using a caliper, if i move 20mm and go back 0.2, i get 19.8mm, so no backlash..

Printed the same testcube without rotating it 45 degrees, and her comes the funny part, spot on..
comes out at 40.15 x 40.15, the other print gave me 39.99 on the "red" side and 40.19 on the yellow side.

So, i still have some kind of problem, to bad there isn't a M579, or is there some way of tweeking it in Marlin ?

I know x and y steps should / must be equal on a corexy, but what happens if i change one of them ?

easy to re calc the x and y to get it down to 40mm from 40.15, but the main problem will still be in the background :)

thinkyhead, 2018-05-10:
Check your G-code to see that it is producing a cube in the correct dimensions. It may be expecting a wider outer wall.

closed 2019-05-16 by thinkyhead