MarlinFirmware/Marlin #10660: [Bugfix-1.1.x] Enable UBL compile error 'ARG_X' has not been declared

Trught, 2018-05-08:
### Description
[Bugfix-1.1.x] 
After enable UBL on stock config compile error.
tested Arduino 1.8.2 & 1.8.5 & 1.9-beta

#define AUTO_BED_LEVELING_UBL
#define RESTORE_LEVELING_AFTER_G28  //UBL condition
#define EEPROM_SETTINGS  //UBL condition

Error:
```
In file included from sketch\stepper.h:46:0,

                 from sketch\stepper.cpp:56:

planner.h:496:42: error: 'ARG_X' has not been declared

     FORCE_INLINE static void buffer_line(ARG_X, ARG_Y, ARG_Z, const float &e, const float &fr_mm_s, const uint8_t extruder, const float millimeters = 0.0) {

                                          ^

planner.h:496:49: error: 'ARG_Y' has not been declared

     FORCE_INLINE static void buffer_line(ARG_X, ARG_Y, ARG_Z, const float &e, const float &fr_mm_s, const uint8_t extruder, const float millimeters = 0.0) {

                                                 ^

planner.h:496:56: error: 'ARG_Z' has not been declared

     FORCE_INLINE static void buffer_line(ARG_X, ARG_Y, ARG_Z, const float &e, const float &fr_mm_s, const uint8_t extruder, const float millimeters = 0.0) {

                                                        ^

planner.h:537:46: error: 'ARG_X' has not been declared

     FORCE_INLINE static void set_position_mm(ARG_X, ARG_Y, ARG_Z, const float &e) {

                                              ^

planner.h:537:53: error: 'ARG_Y' has not been declared

     FORCE_INLINE static void set_position_mm(ARG_X, ARG_Y, ARG_Z, const float &e) {

                                                     ^

planner.h:537:60: error: 'ARG_Z' has not been declared

     FORCE_INLINE static void set_position_mm(ARG_X, ARG_Y, ARG_Z, const float &e) {

                                                            ^

sketch\planner.h: In static member function 'static void Planner::buffer_line(int, int, int, const float&, const float&, uint8_t, float)':

planner.h:500:22: error: 'rx' was not declared in this scope

       buffer_segment(rx, ry, rz, e, fr_mm_s, extruder, millimeters);

                      ^

planner.h:500:26: error: 'ry' was not declared in this scope

       buffer_segment(rx, ry, rz, e, fr_mm_s, extruder, millimeters);

                          ^

planner.h:500:30: error: 'rz' was not declared in this scope

       buffer_segment(rx, ry, rz, e, fr_mm_s, extruder, millimeters);

                              ^

sketch\planner.h: In static member function 'static void Planner::set_position_mm(int, int, int, const float&)':

planner.h:541:24: error: 'rx' was not declared in this scope

       _set_position_mm(rx, ry, rz, e);

                        ^

planner.h:541:28: error: 'ry' was not declared in this scope

       _set_position_mm(rx, ry, rz, e);

                            ^

planner.h:541:32: error: 'rz' was not declared in this scope

       _set_position_mm(rx, ry, rz, e);

                                ^

exit status 1
'ARG_X' has not been declared
```

### Steps to Reproduce

1. Download [Bugfix-1.1.x]
2. Extract Marlin folder to arduino project folder
3. Edit Configuration.h
    #define AUTO_BED_LEVELING_UBL
    #define RESTORE_LEVELING_AFTER_G28
    #define EEPROM_SETTINGS
4. Open in Arduino & Verify
5. Error

Did I miss any other config?
Without UBL is Verify OK.

Roxy-3D, 2018-05-08:
You can turn off ARC_SUPPORT in configuration_adv.h to remove this.   The problem should be fixed tonight.

thinkyhead, 2018-05-09:
Patched!

commit a225d2e9 (2018-05-09): Fix compilation with UBL and Arc/Bézier

Fix #10660
commit b9e4ce37 (2018-05-09): Fix compilation with UBL and Arc/Bézier

Fix #10660
closed 2018-05-09 by thinkyhead