MarlinFirmware/Marlin #10661: Eeprom CRC Mismatch with PIDTEMPBED activated on Anet A8

chrunda, 2018-05-09:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-05-09:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thinkyhead, 2018-05-10:
> [can I] use my already configured files from stable version 1.1.8?

Drop them in place, try to compile. The compiler will throw errors, telling you to change some things.

chrunda, 2018-05-10:
Thank you, but I've rather reconfigured new files with my old values and everything works fine now. No more EEPROM CRC mismatch. I get some EEPROM version mismatch, but I've managed it by M500 and M502. Now I am thinking, where can I save some bytes in code, because I want to use ADVANCED_NOZZLE_PARK function and I am about 5000 bytes over maximum code size. Can you give me some advice?

thinkyhead, 2018-05-10:
5000 bytes is a lot to whittle down. Make sure all debug features like `DEBUG_LEVELING_FEATURE`, `PINS_DEBUGGING`, and so on are disabled. Turn off the `LCD_INFO_MENU`. And try `SLIM_LCD_MENUS` if not already enabled.

closed 2018-05-10 by thinkyhead

chrunda, 2018-05-10:
Thanks for advice, I will try it in the evening.