MarlinFirmware/Marlin #10664: Support for this....

Hebign, 2018-05-09:
Could it be possible to implement support for this :

https://cults3d.com/fr/mod%C3%A8le-3d/outil/filament-tracking-system

Thanks.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-05-09:
We would definitely welcome an implementation submitted by the makers of this unit.

Hebign, 2018-05-10:
For that to be possible, it would already be necessary that the parameters of Marlin allow it, here is what it is necessary to activate in Repetier so that it works:

 Paramètres (firmware Repetier 0.92.9):

 

Jam detection pin : Broche du fin de course sur lequel est connecté le module de détection.

JAM_STEPS : Nombre de pas moteur théorique entre deux signaux (3,6 x nombre de pas par mm de l'extrudeur)

JAM_SLOWDOWN_STEPS : Nombre de pas entre deux signaux au-delà duquel l'impression va être ralentie.

JAM_SLOWDOWN_TO : % de ralentissement de l'imprimante.

JAM_ERROR_STEPS : Nombre de pas entre deux signaux au-delà duquel on considère qu'il y a une anomalie importante (colmatage, fin de bobine...)

JAM_MIN_STEPS : Nombre de pas minimum avant de prendre en compte un nouveau changement de signal. (Permets d'avoir un signal "propre", en passant la zone où peut se produire un éventuel "balbutiement" au changement d'état du capteur")

JAM_ACTION : Choix de l'action à effectuer quand un "bourrage" est détecté.

Hebign, 2018-05-10:
https://www.genapart.com/accueil/infos-d%C3%A9tecteur/

thinkyhead, 2018-05-10:
> For that to be possible, it would already be necessary that the parameters of Marlin allow it

Well, specifically I mean they would need to write the whole implementation, including those (or similar) parameters, the pin assignments, the configuration option to enable the sensor, and any LCD items that ought to go along with it. I would write the implementation, but I have too much else going on right now.

GenaUser, 2018-05-11:
This module is connected to a "signal" pin and a ground / 5V power supply, like the end-stop sensors.
You can read the description of this Repetier firmware option at the section "Jam and out of filament detection" on this page:  [https://www.repetier.com/documentation/repetier-firmware/rf-installation/](url)

boelle, 2019-02-20:
@Hebign try and ask the makers of the unit to submit a PR against bugfix 2.0

Hebign, 2019-05-14:
@boelle what's a "PR" please ?

InsanityAutomation, 2019-05-14:
None needed. Lulzbot already implemented support for this style system awhile ago. Take a look at

// Enable this option to use an encoder disc that toggles the runout pin
    // as the filament moves. (Be sure to set FILAMENT_RUNOUT_DISTANCE_MM
    // large enough to avoid false positives.)
    //#define FILAMENT_MOTION_SENSOR

InsanityAutomation, 2019-10-15:
@boelle @shitcreek this is also done

closed 2019-10-15 by boelle