MarlinFirmware/Marlin #10665: Anet A8 - IS there a way to use any pin from the unused j3 connector to ass a thermistor?

dboa, 2018-05-09:
My heatbed thermistor isn't working properly, is there a way to wire the thermistor trough the unused J3 connection from my anet A8 and change the code on the Marlin Firmware?

Thank you all in advance.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Bob-the-Kuhn, 2018-05-09:
I don't think so.

**IF** I'm reading the correct schematic, all the analog inputs are already used.  The only possibility is if you're not using a LCD.  There are analog inputs on that connector.

I'm looking at a schematic for the Anet 3D/A6/A8 board.  At least that's what the comments say.

Any chance that the problem is a bad thermistor or a bad connection?

dboa, 2018-05-09:
I tried to resolder the resistor, and it gets burned when I try to use it,
If I exchange the termistors, both of them show correcty as the extruder termistor.

I'm going to try to use it without the LCD, since I'm using OCTOPI, it might be OK.

Thank you for the input...

dboa, 2018-05-09:
@Bob-the-Kuhn correct me if I'm wrong, but acording to the Anet A8 Board LCD 2004 schematics:
https://github.com/ralf-e/ANET-3D-Board-V1.0/blob/master/ANET3D_LCD2004_board.png
Isn't the ext-a2 pin unused? Wouldn't I be able to use it for the thermistor?

Bob-the-Kuhn, 2018-05-09:
What resistor are you working with?  Is it the 4.7K on the Anet board?

Sounds like there is a wiring problem.  Does the resistor burn up if the cable is removed?  

Maybe remove the cable and attach a thermistor directly to the board.

dboa, 2018-05-09:
@Bob-the-Kuhn I put the thermistor directly to the board.

Looks like the capacitor is burned as well, maybe that's whats burning the resistor (4.7K)?

Bob-the-Kuhn, 2018-05-09:
Yes, the cap needs to be at least removed.

I doubt that 5V is enough to cause the damage.  You may have stepper motor power shorted to that pin in the cable.

closed 2019-05-16 by thinkyhead