MarlinFirmware/Marlin #10666: ESP8266 + rampsFD V 1 - help to configure (not woking dual serial)

rockman256, 2018-05-09:
Hi!
When in config:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1     
Work USB connection, bot not work ESP8266.
Esp connected to D18 and D19 port.
And when i change
#define SERIAL_PORT -1
#define SERIAL_PORT_2 -1
ESP8266 work is fine, but i can't connect to USB.      
What i need to do? I want use wifi and usb connections.

rockman256, 2018-05-11:
Forget add:
When config is 
#define SERIAL_PORT 1
#define SERIAL_PORT_2 -1
and i plug in SD card at discount smartramps lcd -  system is visibe it as USB disk. But can't connect by serial port.

thinkyhead, 2018-05-25:
DUE only supports a single serial host interface.

boelle, 2019-02-20:
@thinkyhead maybe we can close this one?

closed 2019-05-16 by thinkyhead