MarlinFirmware/Marlin #10671: PROBE_SELECTED not set when enabling BLTOUCH

Thelvaen, 2018-05-10:
in Conditionals_LCD.h the 
#define PROBE_SELECTED is not enabled if BLTOUCH is defined.

Hence HAS_BED_PROBE is not set, and thus the Z_PROBE_OFFSET_FROM_EXTRUDER is not used.

thinkyhead, 2018-05-10:
Seems fine to me:
```cpp
  #if ENABLED(BLTOUCH)
    #ifndef Z_PROBE_SERVO_NR
      #define Z_PROBE_SERVO_NR 0
    #endif
  . . .
  #endif

  #define HAS_Z_SERVO_PROBE (defined(Z_PROBE_SERVO_NR) && Z_PROBE_SERVO_NR >= 0)
  #define HAS_BED_PROBE (ENABLED(FIX_MOUNTED_PROBE) || ENABLED(Z_PROBE_ALLEN_KEY) || HAS_Z_SERVO_PROBE || ENABLED(Z_PROBE_SLED) || ENABLED(SOLENOID_PROBE))
```
As you can see…
`BLTOUCH` => `Z_PROBE_SERVO_NR` => `HAS_Z_SERVO_PROBE` => `HAS_BED_PROBE`.

Thelvaen, 2018-05-10:
My bad, I did not saw that part.

closed 2018-05-13 by thinkyhead