MarlinFirmware/Marlin #10672: reverse_pass_kernel broken

hoffbaked, 2018-05-10:
### Description

The reverse planner kernel is only recalculates if the entry speed is different from the maximum entry speed. If the next block has been recalculated, it should always be recalculated. Currently, if a block can't meet it's deceleration, it isn't getting propagated backward, and bad things are happening. It should look something like this:

void Planner::reverse_pass_kernel(block_t* const current, const block_t * const next) {
  if (!current || !next) return;
  // If entry speed is already at the maximum entry speed, no need to recheck. Block is cruising.
  // If not, block in state of acceleration or deceleration. Reset entry speed to maximum and
  // check for maximum allowable speed reductions to ensure maximum possible planned speed.
  float max_entry_speed = current->max_entry_speed;
  if (current->entry_speed != max_entry_speed || TEST(next->flag, BLOCK_BIT_RECALCULATE)) {
    // If nominal length true, max junction speed is guaranteed to be reached. Only compute
    // for max allowable speed if block is decelerating and nominal length is false.
    float old_entry_speed = current->entry_speed;
    current->entry_speed = (TEST(current->flag, BLOCK_BIT_NOMINAL_LENGTH) || max_entry_speed <= next->entry_speed)
      ? max_entry_speed
      : min(max_entry_speed, max_allowable_speed(-current->acceleration, next->entry_speed, current->millimeters));
    if (current->entry_speed != old_entry_speed) SBI(current->flag, BLOCK_BIT_RECALCULATE);
  }
}

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-05-10:
It appears the suggested change will result in fewer cases where `BLOCK_BIT_RECALCULATE` gets set, so it should result in less recalculation, which would be good.

Has Marlin _always_ had this issue, or is it only in recent versions?

hoffbaked, 2018-05-10:
It would stand to reason if next->entry_speed could have changed, it should always recalculate the entry speed of the current block.

thinkyhead, 2018-05-10:
I see… So the only "less" part that comes from the addition of…
```
const float old_entry_speed = current->entry_speed;
. . .
if (current->entry_speed != old_entry_speed) SBI(current->flag, BLOCK_BIT_RECALCULATE);
```
…won't reduce calculation as much as the `TEST(next->flag, BLOCK_BIT_RECALCULATE)` adds on.

thinkyhead, 2018-05-10:
Ok, I've whipped together some PRs based on your sage advice. Please look them over and see that they're aligned with your observations.

In them I also added the check for a change in `entry_speed` to the first (no 'next' block) test in `reverse_pass`. It seems sensible, but maybe there's some formal reason that the newest block should always get this flag.

ejtagle, 2018-05-10:
This makes some sense. If the "next" block has changed its entry_speed, that means the "current" block has changed its exit speed. And if that is true, it makes a lot of sense to recalculate it.
But, comparing this code to the latest GRBL there is another extra bug... ;)

thinkyhead, 2018-05-10:
> comparing this code to the latest GRBL there is another extra bug

I look forward to learning more about that, and await your PR with bated breath.

closed 2018-05-10 by thinkyhead

ejtagle, 2018-05-10:
We are all learning here... Me too ... ;)

thinkyhead, 2018-05-10:
If only we could find some motion control PhD's to throw in their 2¢.

ejtagle, 2018-05-10:
@thinkyhead : Main problem is that we don´t have the computing power for the algorithm PhD´s would require!
That is why i really like Marlin: The challenge to make everything fit in a small MCU and make it work in realtime.
If you like a computationally heavy algorithm, Klipper could be the answer. But i am pretty sure there is no need for such algorithm... ;)

thinkyhead, 2018-05-10:
@ejtagle — Someone on the Prestigious Internet suggested doing both junction deviation and jerk limiting, combined. First perform junction deviation, then apply jerk limit. I'm not sure how complementary they would really be.

ejtagle, 2018-05-10:
Well, essentially, 2 ways (i am not sure if they are complementary or not) of handling speed transitions between movement segments. Junction deviation plays with the idea of deceleration on sharpo corners, to reduce tangencial acceleration.
Jerk is the opposite, trying to allow instantaneous jumps of speeds between segments.

So, junction deviation slows down printing, in an attempt to reduce tangential acceleration
Jerk control, as Marlin uses, does exactly the opposite, allowing instantaneois jumps in speed, to speed up printing

My thinking on this is that Junction deviation has more "physical meaning", and Jerk control is more like a "hack" (let´s call it "compromise" so noone gets hurt) to speed up printing slightly reducing quality.

The main problem with the proposed Junction deviation is that is unable to "see" more than 2 segments of movement. That is why it fails to properly estimate the tangencial acceleration when segments are small.

If we find a way to do that estimation, then probably Junction deviation is the way to go. And as far as i know, we have the motion planner queue that can be used to estimate it. Maybe using the fact that slicers tend to break rounded corners in small segments, so we can somehow detect the rotation angle by accumulating rotation angles from more than 2 segments (a sliding window comes into mind)...

thinkyhead, 2018-05-10:
It would help if we could encapsulate the planner in a simple C++ program, feed it G-code, and do lots of careful analysis. Of course, this is also possible by adding a debug block to Marlin that turns off real stepping and just prints out the entry/exit speeds for blocks as they would be processed by the trapezoid generator. Standalone would be preferable for the faster build/turnaround time, and being able to run the analysis on a desktop.

ejtagle, 2018-05-10:
That was exactly the thing i was working on before all this jerk/junction algorithm came.
My idea was to better isolate the planner of the stepper classes. And i mostly succeeded, but, that was a big PR, and until we find a workaround or a fix to this problem, i am reluctant to ask for a PR.

The reason i was doing that, was to solve the "stepper pulse jitter" problems the current code has. And to be able to do that, i needed proper encapsulation and not the cross dependency between the Stepper and Planner classes.

You can take a look at my Work in progress at https://github.com/ejtagle/Marlin.git
I am tracking all changes to Planner and Stepper classes.
I still have a small bug related to endstops, so it is not ready to be merged. Hopefully i´ll fix it today 👍