MarlinFirmware/Marlin #10679: Marlin 2.0.x re_arm bugs

cdedwards, 2018-05-10:
After receiving my new re-arm board, I cloned a fresh copy of Marlin 2.0.x and proceeded to program my board for my printer. It's so unstable I can barely do anything with it before loosing the serial connection. Writing to the SD card is a real exercise that fails to put anything on disk even when it says it's completed. Sometimes, I'll get lseek(), write errors, but even if I don't the files on the SD card a garbage. Writing a new kernel to the SD will usually succeed. but loading it leaves me with a blank screen. I have tried multiple SD cards, and USB cables with the same results. I find I have better results with smoothieware. The only way I can update the printer's Marlin, is to remove the SD card, put it directly in the computer.. there is no way to update writing to the SD from Marlin.

M666 X4
M500
will usually cause the serial connection to drop taking out the SD connection as well. When it does work, a M503 is guaranteed to cause it to crash the serial connection. Quite often I will see lost characters, which shows as some misplaced chars from a previous line showing up in the next line.

I get this error quite often

Attempting connection at \\.\COM4...
  Testing plaintext communication protocol...
READ: echo:Unknown command: "XYZv3/query=aG90"
READ: ok
  Connected to machine!
SENT: M666 X4
READ: ok
SENT: M105
READ: ok T:18.571428 /0.000000 B:21.200001 /0.000000 @:0 B@:0
SENT: M105
READ: ok T:18.571428 /0.000000 B:21.200001 /0.000000 @:0 B@:0
SENT: M500
READ: echo:Settings Stored (705 bytes; crc 6201)
Settings Stored (705 bytes; crc 6201)
READ: ok

after M503 crashed I reconnected to receive this in my log.
M421 I8 J8 Z0.000000
READ: echo:Endstop adjustment:
Endstop adjustment:
READ: echo:  M666 XZ<tower angle corrections>
M666 XZ<tower angle corrections>
READ: echo:  M665 L360.320007 R158.384995 H397.552856 S160.000000 B115.000000 X4.000000 Y0.000000 Z0.000000

second attempt for M503
Endstop adjustment:
M666 X0.000000 Y0.000000 Z0.000000

Notice that X is not 4 at all. even after a restart. X is not 4.0 as it should be.

cdedwards, 2018-05-10:
some more info. When I purchased my re-arm, i also purchased a viki2 display. When I disabled the Viki2 by  removing the cable, M500 now works. M666 still does not. I'm still getting serial disconnects, but at least I can save eeprom.dat to disk. I still could not update firmware.bin by copying it to the SD card from the host. I have to physically remove the card and copy the file to it directly. I also shortened up the cable to the Viki2 as the cable that comes with it is ridicously long at 24 inches.

thinkyhead, 2018-05-10:
Sounds like there may be some kind of short involving the serial lines.

p3p, 2018-05-11:
The USB connection does appear to be sensitive to something, .. I don't know what, its not a common problem, a noisy ground plane is known to cause issues for USB. I flash new firmware a lot and have never had a transfer issue, also tested the USB serial with a considerable amount of data and not dropped a character.

I've been putting a fair bit of time into testing the USB stack since the first issue involving disconnects but haven't been able to reproduce the problem although I am working on a new implementation ( can't figure something out start again ;) ) while cleaning up the LPC176x platform, I'l see where that goes.

cdedwards, 2018-05-11:
thank you for looking into this. itès very frustrating. Ièm very sure itès not a electrical short on the serial lines as i am currently running smoothieware and not having any major issues. i have had a few disconnects, so noise could be an issue. i will see about adding a ground plane as i have 3 switching power supplies. 24volt, 12v and a 5v. buck converterès are noisy so hopefully that might resolve some issues. i have put my oscilloscope on the power converterès and they donèt appear noisy. 

sorry about the french chars. not sure why my keyboard is suddenly spewing them

556duckvader, 2018-05-15:
I had issues with cheap USB cables.  I purchased a short quality USB cable and that resolved my connection issues.

cdedwards, 2018-05-15:
how short is short. my computer and printer are not very close. So 3ft cable. I've tried three differant ones. Biggest issue though is not being able to write to the internal SD card. 1 out of four writes to the card fail. I have tried multiple cards. This is just using M500. I get multiple lseek() errors so I  know it fails. trying the save again will eventually work.

boelle, 2019-02-20:
@cdedwards is this firmware or hardware bugs?

boelle, 2019-03-17:
i cant reproduce this and i have 2 re-arms with different usb cables on them

closed 2019-03-17 by p3p