MarlinFirmware/Marlin #10680: Athorbot not complying with License

boelle, 2018-05-10:
i wanted to help someone with their Athorbot Buddy Couple printer and asked the company to send the source code but they refuse and only gave me an hex file which dont do much good as there is no access to the configuration.h and configuration_adv.h

i used athorbot3d@gmail.com address to contact them

FernandoVelcic, 2018-05-10:
I have the same problem with 3dparts/3dmatik (contacto@3dparts.com.ar), they dont want to share to me the configuration.h and configuration_adv.h, they think that a modification in this files without share it do not violate the license.

boelle, 2018-05-10:
it is... even if they dont make a single comma change they have to share

boelle, 2018-05-10:
@FernandoVelcic btw do you have a direct link to the printer they sell?

boelle, 2018-05-10:
the is a reason for the license, and that is to ensure that what started out as open source stays open

and there is a benefit for the makers to share the config files, that is that it will be included with marlin and the maker can just point directly to the marlin project instead of their own site

and then their customers will benefit from all the updates and bug fixes. and then if they give a small money contribution i think the code guy @thinkyhead will be happy too.

FernandoVelcic, 2018-05-10:
@boelle at this time they do not have it published on their website 3dparts.com.ar but I can affirm that they are the same people who sell it in mercadolibre

boelle, 2018-05-10:
that is the most important, having the direct contact to the seller :-), but yes again, they have to share even if they are just changing the 2 configuration files, they are a part of the firmware, without them you cant compile the firmware

boelle, 2018-05-13:
and if the companies dont use that?

if is not flashed on the board it cant spit out anything

best thing is to put pressure on those companies and tell them they are very wrong in what they do and what benefits it has to share the configs

Roxy-3D, 2018-05-13:
>  Should we implement some gcode that would spit out every config and define?

We have been heading in that direction.   Now at power up, the firmware spits out most of the important settings.    And we seem to have gone a little bit too far in that direction by having the mesh based bed leveling systems print out their mesh data.   For the purposes of letting somebody get their settings moved to a newer (non-company supplied) version of the firmware, that seems like over kill.  Especially since the mesh data is stored at the end of the EEPROM and survives across firmware versions.

boelle, 2018-05-13:
Agreed :-) and you could also get the mesh by probing again

and as i just said it does not help anything if the feature is not flashed to the board, or if the "crook"/company disables it

i still think that appling light pressure and telling the benefits of sharing is the way to go

thinkyhead, 2018-05-14:
> best thing is to put pressure on those companies

http://www.3dprintergploffenders.com/

FernandoVelcic, 2018-05-15:
Continuing with the idea of @psavva, I think it can be implemented with a gcode that writes a long string on serial output or sd file with a format like this:

Serialization example
VVVV(marlin release version) defined(DEF1) defined(DEF2) value(DEF3) ...
0118104000.4 ...

Deserialization example:
Marlin version: 01.18
#define DEF1
//#define DEF2
#define DEF3 4000.4

After generate the long string a new web service can take this as input, parse it and generate the all configurations files (configuration.h and configuration_adv.h) with all defines and comments like the original files but with the parsed values.

thinkyhead, 2018-05-24:
> After generate the long string a new web service can take this as input, parse it and generate all configurations files (configuration.h and configuration_adv.h) with defines and comments like the original files but with the parsed values instead.

Not bloody likely. The number of configuration options and the pace of change make this concept extraordinarily difficult. And it wouldn't fit onto 128K boards without sacrificing features. I think this is a non-starter.

boelle, 2018-05-29:
and they can always remove it from the code so the user cant get the config.

The only way is to convince them of the benefits of sharing

closed 2018-06-03 by boelle