MarlinFirmware/Marlin #10681: 'class Temperature' has no member named 'print_heaterstates'

AcrimoniousMirth, 2018-05-10:
### Description

As in title, running 1.1.8.
Full printout:
`sketch/Marlin_main.cpp: In function 'void gcode_M109()':
Marlin_main.cpp:7799: error: 'class Temperature' has no member named 'print_heaterstates'
       thermalManager.print_heaterstates();
                      ^
In file included from sketch/MarlinConfig.h:38:0,
                 from sketch/Marlin.h:36,
                 from sketch/Marlin_main.cpp:247:
sketch/Marlin_main.cpp: In function 'void gcode_M145()':
Marlin_main.cpp:8125: error: 'HEATER_0_MAXTEMP' was not declared in this scope
         lcd_preheat_hotend_temp[material] = constrain(v, EXTRUDE_MINTEMP, HEATER_0_MAXTEMP - 15);
                                                                           ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h:95:60: note: in definition of macro 'constrain'
 #define constrain(amt,low,high) ((amt)<(low)?(low):((amt)>(high)?(high):(amt)))
                                                            ^
exit status 1
'class Temperature' has no member named 'print_heaterstates'
`

thinkyhead, 2018-05-10:
Sounds like you might be using thermocouples on both extruder and bed, or a thermocouple and no heated bed.

Since 1.1.8 will eternally remain as it is, please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

AcrimoniousMirth, 2018-05-10:
@thinkyhead So this arises from the setting of the thermistor in config? I've set mine to 0 as I'm intending to use Marlin as a CNC firmware, which I read was now a supported mode (this suggests otherwise). Cool, quick check here will just be to swap to a different thermistor number and try again.
If you're saying I can't run the firmware without any thermistors I suppose the workaround is to insert a resistor or wire jumper.
Thanks, I'll try that in the morning and report back.

thinkyhead, 2018-05-10:
You may be able to use `bugfix-1.1.x` which has many changes since 1.1.8. And, if it doesn't work, it's a version we can actually patch.

boelle, 2019-02-20:
@AcrimoniousMirth did you try bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead