MarlinFirmware/Marlin #10682: FAST_PWM_FAN does not increse Freq on Pin4

lienbacher, 2018-05-10:
### Description

I have an Anet A8 with Ramps 1.4 and have recently modified it to run a 24V PSU. I have also made a small breakoutboard with a 12 voltage regulator, 3 mosfets and 6 plugs (3 of which controlled by the mosfets, 3 always on) to run auxiliary 12V devices from. This works fine. 

I have configured marlin to use Pin 4 for the part cooling fan, Pin 5 for the heatbreak fan, and pin 6 for the controller fan, all of which works fine. However, I'd love to use speed control on the partcooling fan, which also works fine with my mosfet. However, the 1KHz noise of the fan is louder than everything else and I wanted to increase the pwm frequency using `#define FAST_PWM_FAN` in configuration.h.

However, the noise is the same, and when I measure the pulse with my oscilloscope I get just below 1Khz with FAST_PWM_FAN defined. 

### Steps to Reproduce

1. configure pins_RAMPS.h to use Pin 4 for FAN_PIN
2. uncomment  `#define FAST_PWM_FAN` in configuration.h.
3. run the printer, set a speed between 0 and 255 to get pwm pulses from pin 4 and measure with oscilloscope.

**Expected behavior:** 

I am expecting a higher frequency than 1Khz at pin 4.

**Actual behavior:** [What actually happens]

I still get 1Khz of PWM frequency from pin4

#### Additional Information

I am using latest 1.1.x bufix from today.

thinkyhead, 2018-05-10:
You didn't attach your configurations.

When `FAST_PWM_FAN` is enabled, 7.8KHz is the PWM frequency. When `FAST_PWM_FAN` is disabled, the native PWM frequency is used. If you're seeing a 1KHz rate, maybe `FAN_SOFT_PWM` is enabled…?

lienbacher, 2018-05-11:
Ah indeed forgot. 
FAN_SOFT_PWM is not enabled/commented out.  

[Marlin_conf.zip](https://github.com/MarlinFirmware/Marlin/files/1994397/Marlin_conf.zip)

lienbacher, 2018-05-13:
I am still lost with this issue. Has anybody been able to reproduce this issue or has another idea of why it would not change frequency?

thinkyhead, 2018-05-13:
With `PINS_DEBUGGING` turned on what does `M43` say about pin 4?

thinkyhead, 2018-05-13:
I just realized, pin 4 is associated with the timer we use for the temperature ISR, and it is configured to always run at ~1kHz, so it cannot be used as a PWM pin at any other frequency.

lienbacher, 2018-05-14:
Ahh that makes sense. Thanks for pointing this out, would have never been able to find out my own. I've swapped pins for part cooling and hotend fan and its fine now.

closed 2018-05-14 by lienbacher

holgin, 2019-03-28:
I have the exact same issue on my board, except the frequency is 488Hz and does not change at all with `FAST_PWM_FAN` enabled and of course `FAN_SOFT_PWM` disabled. 
I am using Pin 9. 
I have checked with both 1.1.9 branch/bugfix 1.1.X branch.

Am I stuck with this frequency? Part cooling fan is really noisy right now :(

AnHardt, 2019-03-28:
If you'd have 'the exact same issue' you could usees the 'the exact same solution' for the 'the exact same reasons' 
If not  - open a new issue and provide the requested information.