MarlinFirmware/Marlin #10684: [2.0.x] Question: safe_delay crashes Archim 2.0 board.

marcio-ao, 2018-05-10:
I am trying to debug some code on the Archim board and I am finding that calling "safe_delay" is causing the board to reset.

Obviously "safe_delay" is used in other places, so I'm perplexed as to why this is happening. Does anyone have any suggestions?

-- Marcio

thinkyhead, 2018-05-10:
When is it called? After setup?

marcio-ao, 2018-05-10:
Currently I am calling it from the IS_SD_INSERTED macro, as I am trying to get USB detection to work. Basically I replaced that macro with a call to a function that calls causes the USB code to check whether a drive is inserted, but the USB initialization requires a delay and the whole board reboots.

I guess I don't really need to do this from the IS_SD_INSERTED macro itself. I'll try making it a global variable and see if I can run the USB drive check in the idle loop without causing the board to reboot.

-- Marcio

marcio-ao, 2018-05-10:
Okay, I just found out that the safe_delay() works fine from the idle() function. So perhaps IS_SD_INSERTED is getting called from somewhere unsafe and I shouldn't do any work in it. I'll go ahead and create a global variable for tracking the USB insertion status and just have IS_SD_INSERTED reference that.

thinkyhead, 2018-05-10:
`IS_SD_INSERTED` is specifically intended to read the pin.

marcio-ao, 2018-05-10:
USB does not have a detection pin, the insertion detection happens in software by writing a bunch of SPI commands to the interface chip. I am trying to write my code in such a way that it presents itself to the rest of Marlin as an SD card. This is why I needing to change the behavior of IS_SD_INSERTED. It's the only way I can notify the rest of Marlin that a USB flash drive has been inserted and that it is alright to read the volume.

thinkyhead, 2018-05-10:
You'll just have to make an alternate `IS_SD_INSERTED` to call a function for your special case. If the USB drive is meant to live alongside another SD card port, then there will be more work still.

marcio-ao, 2018-05-10:
I'm not the one calling IS_SD_INSERTED, the rest of the code does. So if I added a new function, all other code that presently calls IS_SD_INSERTED would need to be modified to call that alternative function. I'm not ready to make such an intrusive change.

I think what you may be getting at is that it is a little bit goofy for me to implement USB and have it pretend to be SD card. That is true. As this code matures, it may be useful to replace Sd2Card with a RemovableMedia base class and have Sd2Card and USBHost be subclasses of that, but for now I am simply writing an "fake" Sd2Card class that talks to the USB code and presents itself as an SD card to the rest of Marlin.

marcio-ao, 2018-05-10:
> If the USB drive is meant to live alongside another SD card port, then there will be more work still.

God no. Something tells me this would be very difficult to make work, and somewhat pointless. There's also the interesting question of what happens if a user plugs in a USB hub to their printer and multiple flash drives to that. Right now, my answer is, if you do this, you should not be surprised it your printer crashes.

thinkyhead, 2018-05-10:
> if I added a new function, all other code that presently calls IS_SD_INSERTED would need to be modified to call that alternative function

See my code sample above.

marcio-ao, 2018-05-10:
Ohh. Sorry, misunderstanding. I already am doing what you had suggested above, almost verbatim:

```
#if ENABLED(USE_USB_STICK)
  #define IS_SD_INSERTED Sd2Card::isInserted()
#elif PIN_EXISTS(SD_DETECT)
  #if ENABLED(SD_DETECT_INVERTED)
    #define IS_SD_INSERTED (READ(SD_DETECT_PIN) == HIGH)
  #else
    #define IS_SD_INSERTED (READ(SD_DETECT_PIN) == LOW)
  #endif
#else
  // No card detect line? Assume the card is inserted.
  #define IS_SD_INSERTED true
#endif
```

So we are actually on the same page. But the reason for this ticket is that `thumb_drive_inserted()` -- in my case, `Sd2Card::isInserted()` -- crashes when a `safe_delay()` is called from within it. It seems like someone is calling IS_SD_INSERTED from an interrupt or something, so I have to be careful what I do inside it.

thinkyhead, 2018-05-10:
Sure enough. The safe usage for `safe_delay` differs a lot from safe usage of `READ(pin)`.

marcio-ao, 2018-05-10:
I've been learning that the safest place to do any work in Marlin is the idle() function. Anywhere else is dicey.

On an unrelated note, one difficulty I have run into using the idle() function is that it gets called *a lot*, so it is problematic if I am polling something via SPI. I end up having to use this pattern quite a bit:

```
{
   static uint32_t next_call;
   if (millis() > next_call) {
    next_call = millis() + 100;
     // Do something that should happen no less than every 100 ms
   }
}
```

It works fine, but using 4-bytes of SRAM per timer seems like a waste. I wonder if it makes sense to implement some sort of scheduler in Marlin, for idle tasks. I guess the difficulty would be to make something that didn't end up using more than 4-bytes per task anyway.

marcio-ao, 2018-05-10:
Anyhow, closing this ticket, since it seems like the problem with safe_delay(), it isn't a bug, just user error.

closed 2018-05-10 by marcio-ao

thinkyhead, 2018-05-10:
> I end up having to use this pattern quite a bit

Best to use `ELAPSED(millis(), next_call)` — even though overflow may be exceedingly rare.