MarlinFirmware/Marlin #10685: [Q] How much is needed to add support for a board?

Phr3d13, 2018-05-10:
I just found that Geeetech has released their board schematics for their stm32 based board, the gtm32pro. https://github.com/Geeetech3D/Smartto the schematic for the board is in this repo, supposedly the source will be added soon too.

thinkyhead, 2018-05-10:
To add support for a new board, you just need to add a new pins file with all the necessary pin assignments. Start by making a copy of `pins_STM32F1R.h` then mod the pins according to the diagram.
[image]

Phr3d13, 2018-05-11:
Ok, I got through all the stepper, the heaters, and the temp pins. I'm stuck with misc functions and LCD stuff. With Misc some of the names don't match up and I don't want to just -1 it `//
// Misc. Functions
//
#define SDPOWER            -1
#define SDSS               PA4
#define LED_PIN            PD2

#define PS_ON_PIN          -1
#define KILL_PIN           -1` 

With LCD, the example has a mix of ## style entries and XY# entries:
`//
// LCD Pins
//
#if ENABLED(ULTRA_LCD)

  #if ENABLED(REPRAPWORLD_GRAPHICAL_LCD)
    #define LCD_PINS_RS         49   // CS chip select /SS chip slave select
    #define LCD_PINS_ENABLE     51   // SID (MOSI)
    #define LCD_PINS_D4         52   // SCK (CLK) clock
  #elif ENABLED(NEWPANEL) && ENABLED(PANEL_ONE)
    #define LCD_PINS_RS         PB8
    #define LCD_PINS_ENABLE     PD2
    #define LCD_PINS_D4         PB12
    #define LCD_PINS_D5         PB13
    #define LCD_PINS_D6         PB14
    #define LCD_PINS_D7         PB15`
Should I keep the define style the same? Or are they interchangeable?

xC0000005, 2018-05-11:
On STM32 processors, the pin-naming style should be P<GPIO device letter><Pin number from 0-15>.  Since it's a 103 chip, it can run on stm32duino (the existing HAL for STM32F103), but you'll need to know more about which STM32F103 it is before you can know what to define the board as for building. Also, check to see if it needs any USB pullup transistors turned on - if so, you'll need to make a variant. If it has a bootloader already in place, you'll need to know the starting vector and so on.

xC0000005, 2018-05-11:
> SD_CMD
> SD_CLK
> SD_DATA2
> SD_DATA1
> SD_CLK
> SD_DATA0
> SD_CD
> SD_DATA3
> Which one of the above is most likely to be SDSS?

Based on this:
https://bbs.espressif.com/viewtopic.php?t=1453 <https://bbs.espressif.com/viewtopic.php?t=1453>

I think it’ll be data3, since it seems to be doing the same sort of thing (in fact, many of your pin mappings should be similar for SD usage).

Phr3d13, 2018-05-11:
Ok, another question. With this code chunk:
```
#if ENABLED(REPRAPWORLD_GRAPHICAL_LCD)
    #define LCD_PINS_RS         PD7   // CS chip select /SS chip slave select
    #define LCD_PINS_ENABLE     PA7   // SID (MOSI)
    #define LCD_PINS_D4         PA5   // SCK (CLK) clock
  #elif ENABLED(NEWPANEL) && ENABLED(PANEL_ONE)
    #define LCD_PINS_RS         PD7
    #define LCD_PINS_ENABLE     PA7
    #define LCD_PINS_D4         PB12
    #define LCD_PINS_D5         PB13
    #define LCD_PINS_D6         PB14
    #define LCD_PINS_D7         PB15
  #else
    #define LCD_PINS_RS         PD7
    #define LCD_PINS_ENABLE     PA7
    #define LCD_PINS_D4         PB12
    #define LCD_PINS_D5         PB13
    #define LCD_PINS_D6         PB14
    #define LCD_PINS_D7         PB15
    #if DISABLED(NEWPANEL)
      #define BEEPER_PIN        33
      // Buttons are attached to a shift register
      // Not wired yet
      //#define SHIFT_CLK 38
      //#define SHIFT_LD 42
      //#define SHIFT_OUT 40
      //#define SHIFT_EN 17
    #endif
  #endif
```
And using the above image for reference, does anyone have any guidance as to what goes where? This is all to new to me to be able to translate names for things from the schematic to the pins file. Maybe a reference chart/cheat sheet I missed?

thinkyhead, 2018-05-13:
> Which one of the above is most likely to be SDSS?

`SPI1 NSS` (`PA4`) perhaps?

thinkyhead, 2018-05-13:
> Ok, another question. With this code chunk…

Just looking at the bottom row of pins on the chart above…

[image]

[image]

eltoozero, 2018-07-06:
Is it an across-the-board option to use “PA0” style numbering? Like can I do it with an “Arduino” based board?

I just created a full ‘nude’ atmega2560 variant in Arduino IDE to untangle pin numbering and expose pins not available on the mega for porting to the (otherwise) proprietary Einstart S, but that required manually hacking fastio HAL to get everything working, would make a way more manageable port without that requirement.

Bob-the-Kuhn, 2018-07-06:
Please open a new issue for this.  Make it much easier to follow.

The 8 bit atmega2560 uses the Arduino D0, D1, … logical pin numbering.

The PA0 scheme is used for the 32 bit boards.

I thought that all the 2560 pins were already defined in fastio.  Please give me an example or two of ones that are not.

When mapping a new board I like to use the M43 commands.

eltoozero, 2018-07-18:
Just a relevant follow-up on this, fastio only applies to core marlin functions, for library supported functions (at least u8glib), with pins on PE2 (D/C) and PE6 (RST) as wired on the Einstart mainboard I needed to provide a modified AVR variant for LCD to function.

closed 2018-11-05 by thinkyhead