MarlinFirmware/Marlin #10687: [bugfix-1.1.x] Marlin - Linear Advance - bowden, delta, no extrusion

TheColonelYoung, 2018-05-10:
### Description
I tried to print with Linear advance v1.5 enabled. I have delta printer based on Kossel mini with bowden extruder. I'm using Simplify3D slicer.

My problem is that extruder is nearly not moving. Moving is very little in both direction. Slowly turn around 30 degrees and back. So result is no extruded filament. When I tried to calibrate K value. Starting at K=0 ending at K=2. I have no filament extruded.

With linear advance disabled I have no problem with printing same model with same settings. I tried to print from SDcard and via USB with baudrate 250 000. Same results.

### Steps to Reproduce

Bugfix-1.1.x branch (May 9th)
I enabled Lineard advance with K = 1.5, using Linear Advance v1.5.
Flash firmware to board.
Add command M900 K1.5 to Simplify3D starting script.
Slicer and firmware settings are below.

### Additional Information

My board is MKS GEN L.
Driver for X,Y,Z motors are TMC2130 (SPI, StealthChop, Interpolation).
Driver for E motor is DRV8825 with 32 microsteps.
Bowden length is 60 cm and cannot be shorter.
Material is PLA, width is 1.75mm.

#### Firmware settings

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 100, 100, 100, 325 }

#define DEFAULT_MAX_FEEDRATE                { 500, 500, 500, 100 }
#define DEFAULT_MAX_ACCELERATION       { 9000, 9000, 9000, 10000 }

#define DEFAULT_ACCELERATION                     3000 
#define DEFAULT_RETRACT_ACCELERATION  3000 
#define DEFAULT_TRAVEL_ACCELERATION     3000 

#define DEFAULT_XJERK                 10.0
#define DEFAULT_YJERK                 DEFAULT_XJERK
#define DEFAULT_ZJERK                 DEFAULT_XJERK 
#define DEFAULT_EJERK                 5.0

#define LIN_ADVANCE_K 1.5

#### Slicer settings

Nozzle diametr 0.4 mm
Extrusion multiplier 0.95
Extrusion width 0.44 mm
Layer Height 0.2 mm
Default printing speed 60 mm/s

Retraction distance 20mm
Retraction restart distance - 0mm
Retraction speed 100 mm/s

Coasting distance - Disabled
Wipe Nozzle - Disabled

5D firmware - Enabled
Relative extrusion distance - Disabled
Allow zeroing of extrusion distance (G92 E0) - Enabled
Firmware supports "sticky" parametrs - Enabled
Only retract when crossing open spaces - Enabled

### Files
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1993425/Configuration.zip)

thinkyhead, 2018-05-10:
See if it helps to set `MINIMUM_STEPPER_PULSE` to 1 or 2.

joris57, 2018-05-11:
Playing around with retraction and lin adv settings this afternoon and I experienced that high retraction values (7mm) and high K (1.0) are incompatible. I had to reduce to 2mm retraction and K=0.5. Your values are WAY higher than that. Try a (much) lower retraction value (like 2mm) and see of this improves anything. Also look at the notes of the impact of lin adv on speed on bowden printers in this article : http://marlinfw.org/docs/features/lin_advance.html (A note on bowden printers vs. LIN_ADVANCE).

TheColonelYoung, 2018-05-11:
Compared to me, your values are very low. Today I tried some calibration and few values of retraction but the best settings was K = 2.7 and retraction 20 mm. With this values I have the best result but I maybe expect better. When I use lower retraction length I have so much stringing.

Bowden have around 60 to 65 cm. Better result I have with `MINIMUM_STEPPER_PULSE 1` then 2, but the difference is very small, nearly nothing.

With setting which I described above I have better results then without Linear Advance. But impact on speed is much higher than I expect. It is 3.5 timer slower.

I will try to lower my values. Because I was tunning my high values.

I appreciate your help.

joris57, 2018-05-12:
I think you should check mechanically where the need for these very high retraction settings come from. Play on the tube path? Inner diameter out of spec? Slipping or skipping extruder (Have you tested that your extruder can handle reliably 20mm retraction @ 100mm/s @ 10000mm/s^2?)? 

I'm running a corexy printer with a 60 cm bowden tube and K=0.4 and 3mm retraction (4mm without lin adv) and no stringing, so K=2.7 AND 20mm retraction is really extravagant.

Phaelz, 2018-05-20:
I got something weird here.

I was using a 1.1.8 bugfix version I downloaded on april. My K was 0.65, the extruder really inverted on every movement of the nozzle.
When the printer is turned on, the host showed M900 K0.65.

I've just updated marlin with the newest bugfix, the host now shows M900 K0.65 R0.00 (this R setting did not exist). And now it seems that with K0.65 nothing happens. It's like the linear advance is not even on.

The code changed? It seems now that the version of lin_advance is back to 1.0 where K values where around 100-200.

boelle, 2019-02-20:
@TheColonelYoung still having issues?

TheColonelYoung, 2019-02-21:
Yes kind of. I upgraded my 3d printer (better bowden, hotend, extruder). Now I have shorter retractions (5-7mm), but linear advance is still unusable for me because of noise and lower speed. When I lower retractions, plastic is stringing at every value of K. With longer retractions is very slow and very loud (with TMC2130 in spreadcycle). So I am not using LA anymore. I have tuned my profiles without it.

boelle, 2019-02-21:
@TheColonelYoung tried bugfix 2.0?

DevilByMoonlight, 2019-02-21:
Forgive me if I'm wrong- I'm not an expert in anyway shape or form, I just tinker and tweak until I get things working - I couldn't help to notice the steps per mm setting on the top of your config for your extruder, if your not using a geared extruder your steps per mm  look to be a bit on the high side for E - for example, my delta is using 3, 2130's for X ,Y & Z  80 steps each  as they are on 20 tooth pulleys and a 4988 for E, I'm using a trigorilla board for the short term, which seems to have the 'step divider' jumpers fixed so E is running @  1/16 stepping, and at that setting my measured & calibrated steps are 96.4 for E, therefor if I was at 1/32 steps I would expect it to be  about double = 192.8, obviously if you are using a geared extruder your steps will be higher in proportion to the gearing...  hope this helps

TheColonelYoung, 2019-02-21:
@boelle Yes i tried 2.0 but 4-5 months ago and there was no difference with LA. Maybe is time to give it next try.

@DevilByMoonlight I was using 32 microstepping and 325 was calibrated value, it was non-geared extruder. Now I am using 16 microsteps with different non-geared extruder and I have aroud 140 E-steps (with same LA issue as before). I have no issue to print object, but print with LA was loud and slow.

I tried geared extruder but it didn't help with LA, maybe it was slower.

boelle, 2019-02-21:
@TheColonelYoung yep, 2.0 is updated almost every day

boelle, 2019-02-21:
@TheColonelYoung maybe close this one and open new issue if problem is there

closed 2019-03-03 by thinkyhead