MarlinFirmware/Marlin #10692: Marlin bugfix 1.1x BLtouch problem

tankdoc, 2018-05-11:
Hi, there!

I made a large 3d printer to print a big size prototype, but I faced problem.

I used BLtouch to verify the angle of the bed, but when I select the 'level bed', the movement is weird.

First, the nozzle goes home and get the bed data, it moves to next X coordinate.

But when goes to next coordinate, it doesn't move and makes awful noise.

To prevent that, I changed endstop settings and other problems.

But well, same problem.

The jog mode can moves it, but level bed function doesn't move.

If someone have a solution, Please anybody help.

Thanks

boelle, 2018-05-11:
is the problem the same with bugfix 2.0 ?

tankdoc, 2018-05-11:
well I dunno the problem occurred at bugfix 2.0, but i think that problem is same.
the program is perfect, but.......
my customization doesn't work.
I installed the BLtouch option to level my large bed.
Also followed the tutorial that the inventor has noticed.
However, the level probing through the X axis doesn't work.

In a nutshell, probing along the Y axis works well, but probing along X axis has error with the second X coordinate. I need to solve this problem.

P.S.
I'm so tired of this.......
till now, i revising my code comparing with the Black widow 3d printer configuration.

thinkyhead, 2018-05-12:
Perhaps your Y max feedrate is too high.

Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as requested in the Issue template) and drop them into your next reply. We'll check them over and see if anything is amiss.

tankdoc, 2018-05-12:
Woah.

I didn't think I could get assistance that quickly.

I really appreciate to everyone.

Just hold up, and I will reply my configuration.h and configuration_adv.h file.

P.S.

This printer has 1200*1200 (mm*mm) bed size and 600 (mm) printable height.

tankdoc, 2018-05-12:
Here. there is my configuration and _adv.h files
[configuration and its adv.zip](https://github.com/MarlinFirmware/Marlin/files/1997289/configuration.and.its.adv.zip)

boelle, 2019-02-20:
@tankdoc did you try what thinky suggested?

tankdoc, 2019-02-20:
@boelle
Yeah. I did it long time ago.
But... It didn't work.
So.........
I pended the project.
Sooner or later, i gonna fix that problem.
P.S. this giant printer(I named it AL. Monger) has been taken too many space to our lab. And, i gonna change it differently. That prototype became so rusty(Linear motion guide) and dusty. After graduating the master student course, i will make it work like bigrep.
Thanks to reply to me.
With fingers crossed,
Tankdoc.

boelle, 2019-02-20:
@tankdoc no problem

just going through the issue list to figre who still has a problem and what issues can be closed

closed 2019-05-16 by thinkyhead