MarlinFirmware/Marlin #10697: K-factor Calibration Pattern generator : unexplained error message for certain input values

joris57, 2018-05-11:
I'm using the K-factor Calibration Pattern generator with following values:

Starting Value for K: 0.5
Ending Value for K: 1.1
K-factor Stepping: 0.1

This causes an error message : Your K-Factor range cannot be cleanly divided. Check highlighted Pattern Settings.

In my math, 0.1 **is** a divisor of (1.1-0.5), so this looks like a bug.

Sineos, 2018-05-11:
The K-Factor Range equals to End minus Start => 1.1 - 0.5 = 1.05
The division of 1.05 by 0.1 equals to 10.5, which is no _clean_ divisor. Except your printer is able to print half lines 😜

comps, 2018-05-11:
> The K-Factor Range equals to End minus Start => 1.1 - 0.5 = 1.05

1.1 - 0.5 = 0.6
0.6 / 0.1 = 6

Sineos, 2018-05-11:
hmm, right 😛 
I'll check.

Sineos, 2018-05-11:
https://github.com/MarlinFirmware/MarlinDocumentation/pull/160

False positive due to rounding error in JS. For JS equals 0.6/0.1 = 6.00000000001

closed 2018-05-13 by thinkyhead