MarlinFirmware/Marlin #10698: [Bugfix 2.0] Last Bugfix does not compile: PI not defined

Bergerac56, 2018-05-11:
The last bugfix (32 bit) 7d78f3476b3a8f6e6640d2e279f7e1d5ef5e1a23 does not compile due to the non definition of PI in G26.cpp

Had to add #define PI 3.141592654 (perhaps not the best place to add it but does the job)

      #if ENABLED(ARC_SUPPORT)
        #define PI 3.141592654
        #define ARC_LENGTH(quarters)  (INTERSECTION_CIRCLE_RADIUS * PI * (quarters) / 2)
        float sx = circle_x + INTERSECTION_CIRCLE_RADIUS,   // default to full circle
              ex = circle_x + INTERSECTION_CIRCLE_RADIUS,
              sy = circle_y, ey = circle_y,
              arc_length = ARC_LENGTH(4);

boelle, 2018-05-11:
what do you use to compile with? not sure if it makes a difference but  if using arduino IDE have you tried ver 1.9.0 ?

ManuelMcLure, 2018-05-11:
I've put in a couple of pull requests to fix this. If you want to fix it yourself just replace `PI` with `M_PI` - that's guaranteed to be defined for all build platforms.

Bergerac56, 2018-05-11:
Thanks. Will do

closed 2018-05-11 by Bergerac56