MarlinFirmware/Marlin #10699: MSG_HEATING  usage inconsistency

filipgoc, 2018-05-11:
### Description

_The M104 gcode implementation currently invokes the heating up message, while the equivalent for bed doesn't. Seems to be a simple oversight since M140 does not have this issue, but I want to double check before submitting any PRs._ 

command | MSG_HEATING usage | comment
--- | --- | ---
M104 set hotend temp | 😕 does use | bad. signals unavailability and prevents other messages from being shown (which may well be the case since the printer goes on processing) 
M109 wait for hotend temp | 👍 does use | good. need to explain why the wait. 
M140 set bed temp | 👍 does NOT use | good. set and move on. no need to disturb. 
M190 wait for hotend temp | 👍 does use |  good. need to explain why the wait. 

### Proposal: 

Get rid of the message in M104.

thinkyhead, 2018-05-13:
I concur! This will be patched shortly.

thinkyhead, 2018-05-13:
One additional thing that will need a fix (soon) is to have the message change if you switch from heating to cooling (for example, from the LCD) while doing `M190 R` for example.

closed 2018-05-13 by thinkyhead