MarlinFirmware/Marlin #10700: [2.0.x] TMC2130 RADDS DUE - comiles error - Build 11th May

Industrial-Spiderman, 2018-05-11:
Form experience of compatibility and to be able to trace better I have taken the vanilla build from today, just changed the board to MOTHERBOARD_RADDS - all compiled OK. Then I switches on the TMC2130 define and switched on some axes. - No further changes. I get a compiler error I do not understand. (I am not an expert in coding, sorry).

I tried changing to an earlier build of the TMC library, but to no avail.

### Description

Arduino: 1.8.5 (Windows 10), Board: "Arduino Due (Programming Port)"

arm-none-eabi-gcc: error: CreateProcess: No such file or directory

exit status 1
Error compiling for board Arduino Due (Programming Port).

So how can I get this away, so that I can continue to configure the latest build?

Thanks for your support. Last time it was excellent.

Ludy87, 2018-05-11:
Try Arduino 1.9.0 BETA
https://www.arduino.cc/en/Main/Software

Industrial-Spiderman, 2018-05-11:
Thanks,

That worked. So I hope the 1.9 Beta will have no further drawbacks.

Let me see...

closed 2018-05-11 by Industrial-Spiderman