MarlinFirmware/Marlin #10704: Dual Z stepper driver and dual Z end stop Issue

G3DBuilder, 2018-05-11:
I have setup the dual Z stepper Drivers and dual Z end stop and found very easy to implement.  However I have a new issue maybe.  When I raise the Z axis manually using Repetier Host the X Axis looses home position and I have to re-home it.  Has anybody had this problem?  As I don't normally do this during printing it may not be a problem. It is just weird and kind of an inconvenience. If I don't re-home the x axis will go past its end limit.

boelle, 2018-05-11:
not an issue here and i have used it for a while, i got lazy about having to sync the 2 z steppers :-D

GMagician, 2018-05-11:
I have dual stepper, dual endstop and repetier host but no issues

fabtopia, 2018-05-12:
the is a thing in the configuration_adv.h with the max_x

//#define Z_DUAL_STEPPER_DRIVERS

#if ENABLED(Z_DUAL_STEPPER_DRIVERS)

  //#define Z_DUAL_ENDSTOPS

  #if ENABLED(Z_DUAL_ENDSTOPS)

    **#define Z2_USE_ENDSTOP _XMAX_**

    #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0

  #endif

#endif

if you presume it is Z2_USE-ENDSTOP _ZMAX_ like de default with X and Y then your machine will do "stange" things

G3DBuilder, 2018-05-18:
The default in configuration_adv.h is to use XMAX so that is what I used.  Are you saying to use ZMAX instead?

thinkyhead, 2018-05-20:
First question we like to ask before anything else is "What version of Marlin are you using?"

_XMAX_ should be fine.

boelle, 2019-02-20:
@G3DBuilder still having issues if you use bugfix 2.0?

G3DBuilder, 2019-02-21:
I am currently using 1.1.9. during daily use this is not an issue. I just brought the issue up in the beginning as I thought it was a little strange. that the X axis would loose home position when ever I raised the Z axis manually. There are no issues during a normal print.

boelle, 2019-02-21:
oki, try and give bugfix 2.0 a try, its both 8 bit and 32 bit

boelle, 2019-03-13:
@G3DBuilder was the issue present in 2.0?

closed 2019-03-17 by p3p