MarlinFirmware/Marlin #10706: problem to set axis to homing position (prusa i3)

makerPeppe, 2018-05-11:
Hi, i'm making a 3d printer: prusa i3.
During the moving of axis at homing position i had a problem that I show it in the photo.
How Can i set the correct home position? 
I apologize for my incorrect English but I speak Italian.

This is part of my code:
```

// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
//#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG

// coarse Endstop Settings
#define ENDSTOPPULLUPS // Comment this out (using // at the start of the line) to disable the endstop pullup resistors

#if DISABLED(ENDSTOPPULLUPS)
  // fine endstop settings: Individual pullups. will be ignored if ENDSTOPPULLUPS is defined
  //#define ENDSTOPPULLUP_XMAX
  //#define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
//#define X_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
//#define Y_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
//#define Z_MAX_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false // set to true to invert the logic of the probe.

---------------------------------------------------------------------------------------------
// @section homing

//#define NO_MOTION_BEFORE_HOMING  // Inhibit movement until all axes have been homed

#define Z_HOMING_HEIGHT 20  // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...
                             // Be sure you have this distance over your Z_MAX_POS in case.

// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR 1 // 1 = posiz home x -> endstop a destra(connessione in ramps su xmax pins) il contrario vale per -1
#define Y_HOME_DIR -1 // -1 = posiz home y -> endstop sul retro (connessione in ramps su ymin pins) il contrario vale per 1
#define Z_HOME_DIR -1 // -1 = posiz home z -> endstop in basso 
// posizione home x= 1; y= -1; z= -1 -> carrello x a destra, piatto y dietro e z in basso

// @section machine

// The size of the print bed
#define X_BED_SIZE 200
#define Y_BED_SIZE 200

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 200

/**
 * Software Endstops
 *
 * - Prevent moves outside the set machine bounds.
 * - Individual axes can be disabled, if desired.
 * - X and Y only apply to Cartesian robots.
 * - Use 'M211' to set software endstops on/off or report current state
 */

// Min software endstops curtail movement below minimum coordinate bounds
#define MIN_SOFTWARE_ENDSTOPS
#if ENABLED(MIN_SOFTWARE_ENDSTOPS)
  #define MIN_SOFTWARE_ENDSTOP_X
  #define MIN_SOFTWARE_ENDSTOP_Y
  #define MIN_SOFTWARE_ENDSTOP_Z
#endif

// Max software endstops curtail movement above maximum coordinate bounds
#define MAX_SOFTWARE_ENDSTOPS
#if ENABLED(MAX_SOFTWARE_ENDSTOPS)
  #define MAX_SOFTWARE_ENDSTOP_X
  #define MAX_SOFTWARE_ENDSTOP_Y
  #define MAX_SOFTWARE_ENDSTOP_Z
#endif
----------------------------------------------------------------------------------
// @section homing

// The center of the bed is at (X=0, Y=0)
//#define BED_CENTER_AT_0_0

// Manually set the home position. Leave these undefined for automatic settings.
// For DELTA this is the top-center of the Cartesian print volume.
//#define MANUAL_X_HOME_POS 196.4
//#define MANUAL_Y_HOME_POS 15.5
//#define MANUAL_Z_HOME_POS 2.4

// Use "Z Safe Homing" to avoid homing with a Z probe outside the bed area.
//
// With this feature enabled:
//
// - Allow Z homing only after X and Y homing AND stepper drivers still enabled.
// - If stepper drivers time out, it will need X and Y homing again before Z homing.
// - Move the Z probe (or nozzle) to a defined XY point before Z Homing when homing all axes (G28).
// - Prevent Z homing when the Z probe is outside bed area.
//
//#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif

// Homing speeds (mm/m)
#define HOMING_FEEDRATE_XY (50*60)
#define HOMING_FEEDRATE_Z  (4*60)
```

[image]
[image]
[image]

boelle, 2018-05-11:
looks like only Y axis is the problem?

change #define Y_MIN_POS 0 to maybe #define Y_MIN_POS -20

when you send command G0 Y0 it should go to the white line

makerPeppe, 2018-05-12:
For me the min position(0) should coincide with the position 15.5 to have the nozzle on the corner of the printer area.
Between the bed and the nozzle there are 15.5 mm of air.
I changed the min-position to 15.5 already but I have not solved the problem.
If I am not able to find a solution in the firmware I will change the length of the smooth bars of my printer

boelle, 2018-05-12:
#define Y_MIN_POS -15.5

it has to be negative as the nozzle is outside the bed

and you say there is alo 15.5 mm of air between nozzle and bed?

makerPeppe, 2018-05-13:
I measured the air between nozzle and bed. 
I have set up the offset with m428 command and uploaded it on the eeprom with  m500 command.
if i move the axis with G0 or G1 command at the initial position (at the home position -> G0 x0 y0 z0) the bed move in the correct home position!
If I, instead, use command g28 to move the axis at the home position the bed come back at the home position without offset. Because it?

boelle, 2018-05-13:
is the missing offset both in Y and Z direction?

boelle, 2018-05-13:
is EEPROM_SETTINGS enabled ? ie #define EEPROM_SETTINGS or is it //#define EEPROM_SETTINGS ??

boelle, 2018-05-13:
the best way is to do as i suggested.... if y is 15.5 mm outside bed then #define Y_MIN_POS -15.5

using offset is kind of cheating

or said in another way, the less you need to use eeprom the better

makerPeppe, 2018-05-13:
No the offset is missing in x and y direction when I set up g28 command while if I set up  the movement of the axis with G0 or g1 command, my printer 3d considers the offset that I have set up. The z offset I resolved moving the z endstop.
I enabled EEPROM_SETTING and the offset that I have set up with m428 command I saved it on the eeprom with the m500 command. I'll try your last solution and I'll write you on it.

boelle, 2018-05-13:
yep... try not to use offset, that is kind of last thing you should do

but remember to remove offset from eeprom or else you will get even more trouble :-D

idanmos, 2018-05-14:
I've a problem with same settings but for Prusa MK42 heatbed, what should be the BED_SIZE vars?

Setting:
```
#define X_BED_SIZE ??
#define Y_BED_SIZE ??
```

Like in Prusa fork:
```
#define X_MIN_POS 0
#define X_MAX_POS 250
#define Y_MIN_POS -2.2
#define Y_MAX_POS 210
#define Z_MIN_POS 0.15
#define Z_MAX_POS 210
```

It cannot be (250, 210) for BED_SIZE because it gives me an errors:

> In file included from sketch/fastio.h:33:0,
>                  from sketch/MarlinConfig.h:26,
>                  from /Users/idan.moshe/Downloads/3D Printing/Firmware/Marlin/Prusa i3 MK2S - Clone/Prusa-i3-MK2-Clone-r1/Marlin-1.1.8/Marlin/Marlin.ino:31:
> Configuration.h:793: error: floating constant in preprocessor expression
>  #define Y_MIN_POS -2.2
>                     ^
> sketch/macros.h:140:32: note: in definition of macro 'WITHIN'
>  #define WITHIN(V,L,H) ((V) >= (L) && (V) <= (H))
>                                 ^
> sketch/Conditionals_post.h:976:27: note: in expansion of macro 'max'
>      #define _MIN_PROBE_Y (max(Y_MIN_BED, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
>                            ^
> sketch/Conditionals_post.h:976:42: note: in expansion of macro 'Y_MIN_POS'
>      #define _MIN_PROBE_Y (max(Y_MIN_BED, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
>                                           ^
> sketch/Conditionals_post.h:986:25: note: in expansion of macro '_MIN_PROBE_Y'
>      #define MIN_PROBE_Y _MIN_PROBE_Y
>                          ^
> sketch/SanityCheck.h:897:40: note: in expansion of macro 'MIN_PROBE_Y'
>    #elif !WITHIN(Z_SAFE_HOMING_Y_POINT, MIN_PROBE_Y, MAX_PROBE_Y)
>                                         ^
> Configuration.h:793: error: floating constant in preprocessor expression
>  #define Y_MIN_POS -2.2
>                     ^
> sketch/macros.h:140:32: note: in definition of macro 'WITHIN'
>  #define WITHIN(V,L,H) ((V) >= (L) && (V) <= (H))
>                                 ^
> sketch/Conditionals_post.h:976:27: note: in expansion of macro 'max'
>      #define _MIN_PROBE_Y (max(Y_MIN_BED, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
>                            ^
> sketch/Conditionals_post.h:976:42: note: in expansion of macro 'Y_MIN_POS'
>      #define _MIN_PROBE_Y (max(Y_MIN_BED, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
>                                           ^
> sketch/Conditionals_post.h:986:25: note: in expansion of macro '_MIN_PROBE_Y'
>      #define MIN_PROBE_Y _MIN_PROBE_Y
>                          ^
> sketch/SanityCheck.h:897:40: note: in expansion of macro 'MIN_PROBE_Y'
>    #elif !WITHIN(Z_SAFE_HOMING_Y_POINT, MIN_PROBE_Y, MAX_PROBE_Y)
>                                         ^
> exit status 1
> floating constant in preprocessor expression

makerPeppe, 2018-05-15:
I resolved setting up Y_MIN_POS to -17.4 and deleting the value of offset that i have set up before.
While for x axis I moved the end stop on it.

boelle, 2019-02-20:
@makerPeppe so problem is solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead