MarlinFirmware/Marlin #10707: Reduce M421 EEPROM Chit Chat to a few lines.

Roxy-3D, 2018-05-11:
With a 10 x 10 mesh...  Or worse, a 15 x 15 mesh...  The M421 values go on and on and on at startup.
The intent of the EPROM CHIT CHAT at the start up is to make it easy to capture settings so they can be propagated to later versions of the firmware.  But 225 lines of M421 is a bit excessive (for a 15 x 15 mesh).

How about we add a new parameter so M421 can absorb an entire row of the mesh matrix?   One idea would be to add an 'R' parameter.   And if that is present it supersedes the old M421 behavior.  If the 'R' parameter is present we then use  'A' through 'P' to specify column values for the row.  (Omitting G and N from use)

If we did something like this, we could collapse the M421 values down to just a few lines but still preserve the intent of having all the printer settings available for inspection.

boelle, 2018-05-11:
i  would say with some carefull thinking we could just print 15 lines with 15 values each

i too have been a bit "annoyed" by the many lines but have adapted to a routine of not using the printer for about 10 mins after each reset... a small pain thou

Roxy-3D, 2018-05-11:
The M421 is important because it allows the text to be sent back to the printer to load the mesh matrix.

boelle, 2018-05-11:
oh i was just thinking about the 225 lines on the serial output at startup... programming wise i'm out of ideas, i know that repetier only displays the mesh +- values and that is 22 lines with 22 values each.

teemuatlut, 2018-05-11:
I think I'd still be bothered by just 10 or 15 unnecessary lines. I don't even like how the `M906` (etc) was divided into multiple lines.
I would rather see a separate command, or a parameter to an existing one, the user can use to retrieve the mesh in gcode format. Just like you can use `G29 T1` for tab separated output.

Roxy-3D, 2018-05-12:
Yeah...  I hear you...   The thing is 'Yet Another Configuration' option would be the 'right' way to cure this.  But we have too many of those already.

Let's take a step back and look at things from a higher level.   Do we really need to preserve the mesh data if somebody is moving to a new level of the firmware?   That is already done intentionally by storing the mesh data at the tail end of the EEPROM to make it inert from EEPROM structure of different firmware revisions.   And if you really care, you can do a G29 S -1 on UBL to save your mesh data.   

Maybe it makes sense to just back off and not print the mesh data on every reset of the board?

Opinions are welcome....

teemuatlut, 2018-05-12:
On my Archim2 (=Due) board the EEPROM is emulated in flash, so every time I need to re-flash the firmware I also lose the mesh data. I do have ways around this to restore it but perhaps it's something that can be taken into account.
It seems `G29 S-1` already does what I was looking for, so maybe we can just drop the output from `M503` altogether?

thinkyhead, 2018-05-13:
Sure, there's no special reason to echo everything on `settings.load`.
I think it's fine to only echo them on an explicit `M503`.

boelle, 2019-02-20:
@thinkyhead should this not get a label?

boelle, 2019-06-22:
missing a label

closed 2019-07-11 by thinkyhead