MarlinFirmware/Marlin #10709: Q: Added dummy stepper to ramps config

boelle, 2018-05-11:
now that i got thermocouple working i wanted to "hack" in support for heated chamber 

i thought: since i have a thermistor input free for a 2nd extruder that i dont use anyway who not use that to take chamber temp readings. i would only need to say i have 2 extruders and add a dummy extruder since i want to use the 2nd  extruder socket for a dual z stepper setup. i added this:

```
#define E1_STEP_PIN        40
#define E1_DIR_PIN         64
#define E1_ENABLE_PIN      59
#ifndef E1_CS_PIN
  #define E1_CS_PIN        63
#endif
```

i tried to use -1 for pin numbers but then compile complained that the stepper was not there, so i figured that any free pin would make do

**Q: is it possible to add a dummy that do not take up any pins?**

i thought why not add a 2nd heater so i dont have to change the wiring and i would just need to change pin numbers once i decide to add a heating element

i added these 2 lines:

```
#define HEATER_1_PIN     42
  #define FAN1_PIN         44
```

in this section

```
#if ENABLED(IS_RAMPS_EFB)                      // Hotend, Fan, Bed
  #define FAN_PIN        RAMPS_D9_PIN
  #define HEATER_BED_PIN RAMPS_D8_PIN
#define HEATER_1_PIN     42
  #define FAN1_PIN         44
```

again i would have liked to not take up any real pins before i knew that a chamber heater is needed

of course my simple thinking did not work

thinkyhead, 2018-05-14:
We already have formal support for a chamber temperature sensor. Just set these options in your `Configuration.h`:

```cpp
#define TEMP_SENSOR_CHAMBER 1
#define TEMP_CHAMBER_PIN    2   // Analog Input
```
…setting `TEMP_CHAMBER_PIN` to whatever analog input your sensor is connected to.

boelle, 2018-05-14:
yeah but that does not output the temp to LCD

thinkyhead, 2018-05-25:
"[FR] Display Chamber Temperature on LCD"

boelle, 2018-05-25:
Hmm i could go with that title. Do we have a FR for full chamber control?

boelle, 2018-07-19:
closing this one  in the hope that one day we have true chamber control and not just a chamber temp on the serial line

closed 2018-07-19 by boelle