MarlinFirmware/Marlin #10712: [bugfix-2.0.x] Due & RADDS

Industrial-Spiderman, 2018-05-11:
### Description

Compiler error if bed sensor is set to anything else than 0

### Steps to Reproduce
I have used a vanilla build of today, changed the board to MOTHERBOARD_RADDS, no other changes made. Compiled OK. 
With a single additional change of the bed sensor definition - #define TEMP_SENSOR_BED 1 the compile fails with following error:

Arduino: 1.9.0-beta (Windows 10), Board: "Arduino Due (Programming Port)"

In file included from sketch\src\module\thermistor/../../inc/../HAL/HAL_DUE/HAL.h:40:0,

                 from sketch\src\module\thermistor/../../inc/MarlinConfig.h:28,

                 from sketch\src\module\thermistor/thermistors.h:26,

                 from sketch\src\module\temperature.h:30,

                 from sketch\src\module\temperature.cpp:27:

sketch\src\module\thermistor/../../inc/../pins/pins_RADDS.h:150:33: error: pasting "/* BED*/" and "_WPORT" does not give a valid preprocessing token

 #define HEATER_BED_PIN      7   // BED

                                 ^

sketch\src\module\thermistor/../../inc/../HAL/HAL_DUE/fastio_Due.h:75:33: note: in definition of macro '_WRITE'

   volatile Pio* port = (DIO ##  IO ## _WPORT); \

                                 ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:939:31: note: in expansion of macro 'WRITE'

   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)

                               ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:939:37: note: in expansion of macro 'HEATER_BED_PIN'

   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)

                                     ^

sketch\src\module\temperature.cpp:928:11: note: in expansion of macro 'WRITE_HEATER_BED'

           WRITE_HEATER_BED(LOW);

           ^

sketch\src\module\thermistor/../../inc/../pins/pins_RADDS.h:150:33: error: pasting "/* BED*/" and "_PIN" does not give a valid preprocessing token

 #define HEATER_BED_PIN      7   // BED

                                 ^

sketch\src\module\thermistor/../../inc/../HAL/HAL_DUE/fastio_Due.h:76:31: note: in definition of macro '_WRITE'

   uint32_t mask = MASK(DIO ## IO ## _PIN); \

                               ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:939:31: note: in expansion of macro 'WRITE'

   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)

                               ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:939:37: note: in expansion of macro 'HEATER_BED_PIN'

[... 299 lines cut ...]

   uint32_t mask = MASK(DIO ## IO ## _PIN); \

                               ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:939:31: note: in expansion of macro 'WRITE'

   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)

                               ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:939:37: note: in expansion of macro 'HEATER_BED_PIN'

   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)

                                     ^

sketch\src\module\temperature.cpp:1833:50: note: in expansion of macro 'WRITE_HEATER_BED'

         if (soft_pwm_count_BED <= pwm_count_tmp) WRITE_HEATER_BED(LOW);

                                                  ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

porcinus, 2018-05-12:
I am using a similar setup with a RADDS board and TEMP_SENSOR_BED set to 1 without any trouble.
The only time I had trouble was when disable PIDTEMP.

I think problems come from your motherboard definition, you need to define MOTHERBOARD to BOARD_RADDS, not MOTHERBOARD_RADDS

Industrial-Spiderman, 2018-05-12:
Thanks,

Sorry my fault, I got BOARD_RADDS in the config - was just my sloppiness describing. I have a copy of 2.0.x end March running just fine, but I wanted to take benefit of some of the newer features implemented since then. Here the config.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1997746/Configuration.zip)

Greetings

p3p, 2018-05-12:
This appears to be the avr-gcc preprocessor bug introduced with Arduino IDE 1.9 beta, until they fix it it can be worked around by removing the comment from the line mentioned in the error (contains `error: pasting `) there may be more than one as any pin definition with a comment on the same line will cause this.

Industrial-Spiderman, 2018-05-13:
Hello and thanks!  That did it - now I can go on.

closed 2018-05-13 by Industrial-Spiderman