MarlinFirmware/Marlin #10724: [bugfix_2.0.x] #error "Sorry! LIN_ADVANCE is only compatible with Cartesian."

dembach, 2018-05-12:
Why is LIN_ADVANCE is only compatible with Cartesian ?
The Extruder, which the matter of K is about linear Advance, is the same ..

can someone clarify, why this is not possible for deltas ?

thanks

Markus from Germany

ejtagle, 2018-05-12:
It is possible for deltas. It is just not implemented right now. we are working on planner cleanup speedup (results are incredible even without LA advance) but after that LA will have a "restyling" (a plain rewrite) to make it compatible with bezier jerk control of speed (already present but you must enable explicitly and does not work well with LA) and Junction deviation (instead of jerk control) is also being integrated.
Printing time reductions of 25% and even better quality (and LA, when rewritten, will improve on that...)
So, please stay tuned 👍

thinkyhead, 2018-05-13:
@dembach — Sure you're not using an outdated version of Marlin?
We no longer have that sanity check in the current code,
and Linear Advance is now compatible with Delta.

boelle, 2019-02-20:
@dembach did you test?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead