MarlinFirmware/Marlin #10725: [bugfix-2.0.x]

Zwaubel, 2018-05-12:
I'm recently experiencing random axis shut offs during prints:
The printer is starting of just fine and in the middle of an arbitrary layer one axis shuts off and is deactivated (so thath I can move the axis by hand) for the rest of the print and the printer is continuing its print job in two dimensions rather than in three.
When canceling the print, the axis comes back to life immediately as if nothing had happened.
It's not always happening, but if the error ocurres, than its specific to one gcode file.
I checked those files but did not find any command that could have caused that behavior.
Interestingly whenever this strange behavior ocurred, the corresponding stl files was sliced by either cura 3.3.1 or slic3r prusa edition. This does not apply the other way around: cura 3.3.1 and slic3r prusa edition don't always produce unprintable gcode.

I checked my electronics twice, recalibrated the currents, improved the cooling... all without any effect.

Machine details:
cartesian printer, RAMPS1.4+Mega2560, no heated bed

GCODE files that caused the issue:
[gcode_files.zip](https://github.com/MarlinFirmware/Marlin/files/1997966/gcode_files.zip)

printer config:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1997967/config.zip)

picture of the treefrog print failing:
[image]

thinkyhead, 2018-05-13:
Is this only occurring in the latest code?
And not occurring with code from —say— three to five days ago?

Zwaubel, 2018-05-13:
I'm struggeling with this issue since about 2 weeks (all based on the bugfix-2.0.x branch). The refactoring of the planner and stepper did not change anything on those random shutdowns.

Maybe it is just time to move to a 32 bit board. The trusty old 8 bit mega really is at the boarder of its computational capacity and with all the upcoming features and ideas for Marlin 2 the mega will probably become a bottleneck for whats possible.
I just can not find a 32 bit board that fits my needs and wishes..

Zwaubel, 2018-05-17:
Unfortunately the problem still remains even with ejtagles planner and stepper optimisations.
Just tried to print another treefrog and the X axis randomly shut off mid print.

Here is the gcode file sliced with cura 3.3.1:
[GRNEO_treefrog_45_cut.zip](https://github.com/MarlinFirmware/Marlin/files/2012159/GRNEO_treefrog_45_cut.zip)

And here is picture of what happened:
[image]

smoki3, 2018-05-17:
I also noticed some error. First i thought my motors a loosing steps but that isnt the issue

thinkyhead, 2018-05-24:
I'm unable to reproduce this issue on my hardware. Random axis shutoffs would tend to indicate faulty stepper drivers. The firmware only shuts down drivers when certain features are enabled, and then only after a period of inactivity. If you turn off all the `DISABLE_INACTIVE_*` features the code to turn off the steppers won't even be compiled in, so you can use that to rule out auto-shutoff.

Zwaubel, 2018-05-24:
@thinkyhead
Thank you for trying, anyways. 
I tuned my printers motion settings and I think therewith fixed its compatibility with gcode produced with slic3r (no steploss or shutdowns on 6+ hour prints). I had to drastically slow down my maximum x and y axes velocity. They are now at 90 mm/s (max), everything higher than that would randomly lead to thunk noises, steploss up until complete axis shutdowns. Playing with the stepper current did nothing but an increase of the  temps of the electronics.
I'm using tmc2208 on x, y and z operating in stealth chop 2.0, and a drv for the extruder. My printer is quite small, 150mm for all axes. Is the atmega 2560 just to weak to enable higher speeds?

thinkyhead, 2018-05-25:
I've been using S3D and haven't used Cura in a while. I'll do some test cubes soon and see how they look with the two slicers.

Zwaubel, 2018-06-02:
Sorry for the delayed response. I had a surgery last week and I'm currently a little handicapped. 

Just found something interesting in the FAQ of the Klipper Firmware:

### My TMC motor driver turns off in the middle of a print

There have been reports of some TMC drivers being disabled in the middle of a print. (In particular, with the TMC2208 driver.) When this issue occurs, the stepper associated with the driver moves freely, while the print continues.

It is believed this may be due to "over current" detection within the TMC driver. Trinamic has indicated that this could occur if the driver is in "stealthChop mode" and an abrupt velocity change occurs. If you experience this problem during homing, consider using a slower homing speed. If you experience this problem in the middle of a print, consider using a lower junction_deviation setting.

Zwaubel, 2018-07-01:
I'm closing this issue for now. Slic3r made GCode works fine, Cura rarely messes up but I'm not using it that often so I can live with that.

closed 2018-07-01 by Zwaubel