MarlinFirmware/Marlin #10730: ABL_BILINEAR_SUBDIVISION with EXTRAPOLATE_BEYOND_GRID: Z coordinate outside probe area invalid

b3nn0, 2018-05-13:
### Description

Hi,
Branch: bugfix-2.0.x
When I have ABL_BILINEAR_SUBDIVISION enabled, together with EXTRAPOLATE_BEYOND_GRID,
there is a weird behaviour when the nozzle moves outside of the probing rectangle on the lower right corner.
For all other corners, extrapolation seems to work fine. But for the lower right corner, the Z coordinate of the nozzle will move to zero, as it can be seen in this video:
https://cloud.batzill.com/index.php/s/K9gy8ZsEM8cTSLF
it does not happen when ABL_BILINEAR_SUBDIVISION is turned off.

### Steps to Reproduce

1. Probe the bed with AUTO_BED_LEVELING_BILINEAR, ABL_BILINEAR_SUBDIVISION and EXTRAPOLATE_BEYOND_GRID enabled
2. Move the nozzle outside the probe area in the lower right corner

Expected behavior: Properly extrapolate the probe grid

#### Additional Information

Config:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1998392/config.zip)

thinkyhead, 2018-05-13:
Very interesting! So, off the grid on the right works, and off the grid in front works, but the two of them together get confused… And it only affects `ABL_BILINEAR_SUBDIVISION` too! I'll look very closely at the code. Something might leap out.

boelle, 2019-02-20:
@b3nn0 is this still a problem with latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead