MarlinFirmware/Marlin #10732: Q: TMC spi and hardware spi share pins

boelle, 2018-05-13:
saw the new video with chris adding tmc2130

i then looked at the pin fiel for ramps and saw that it shares pins with the max chip i just got working

#define TMC_SW_MOSI    66

&

#define MAX6675_SS       66 

are there any reason for this? ie i wonder why it does not use the same spi bus and just add's extra CS pins

Bob-the-Kuhn, 2018-05-14:
The SD card almost always runs on a hardware SPI bus.  Very few other devices also use this bus.  I only know of a couple of LCD displays that do this.

There is no such thing as a software SPI bus, just pins used by software SPI routines.  These routines can use common pins just so long as the chip/slave selects are dedicated.  

In your situation it looks like you have four SPI users:
SD card
LCD display
TMC2130
MAX6675

The LCD & SD card SPIs usually use dedicated SPIs just because the pins are not physically available.

The TMC2130 usually ends up using a software SPI just because the hardware SPI pins are not available.

The MOSI, MISO and SCK pins of the TMC2130 software SPI can also be used for the MAX6675.  Just make sure the chip/slave selects are NOT shared.

boelle, 2018-05-14:
Hehe, even with SD card and LCD on the ramps the MOSI, MISO and SCK pins are still Available, that is how I added the max chip for the hotend. But yes CS pin needs to be dedicated 😆

When I have time I plan to add the 2130`s to the same MOSI, MISO and SCK pins but each will have its own CS pin