MarlinFirmware/Marlin #10733: Possible TMC2130 overheat?

MarioStoilov, 2018-05-13:
I sometimes get a driver error (randomly) when printing. Sometimes it's 30 minutes into the print, sometmes it's 30 hours into the print (DAMN IT!!!!).

The last time I got an error, I looked at the logs (attached as images, because octoprint doesn't handle copy-paste very elegantly) and the only troubling thing I'm noticing is the otpw, which my understanding is that it should not trigger an error, or am I wrong?
[Logs.zip](https://github.com/MarlinFirmware/Marlin/files/1998829/Logs.zip)

thinkyhead, 2018-05-13:
Correct. It shouldn't abort just for optw (at least not in the current bugfix versions). If you disable `STOP_ON_ERROR` then it won't abort for _any_ error.

Be sure to test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch. If you still get freezes with `STOP_ON_ERROR` disabled, then we should look deeper.

marcio-ao, 2018-05-18:
We had so much overheating issues with the TMC2130 that we ended up having to use thermal tape and a metal metal block across the back of our driver chips to conduct heat from the drivers to the controller chassis.

Our experience is that the TMC2130 are certainly the hottest thing in 3D printing right now...

Garyr14, 2018-05-23:
Hello,
I'm relatively new to this, so please excuse my ignorance...  My MKS GEN V 1.4 with TMC2130's on all axis's was working fine until I removed the bed to re-solder the thermistor wire.  Ever since, I've been getting layer shifting on Y axis only.  I've done just about everything troubleshooting wise that I can think of including reverting back to Marlin 1.1.6 (was on 1.1.x bugfix), I've swapped the drivers, wires and motor.  I thought it was the cable catching but that does not seem to be the problem. 

I am curious about the temperature issue.  I've been measuring the temperature using a non-contact digital laser infrared thermometer.  Temps range from 26-33 c max when running.  Motors are lower.  I'm only using heatsinks on the drivers.  I'm curious as to why I'm not having the overheating problems that everyone seems to have with the TMC2130's.  I'm running them between 50 - 70 mms.  Am I taking the temperature right or is there another way to do so?  Again, the only shifting I'm having is on the Y axis.  Everything else is fine.

If there is a more appropriate place to post this, please let me know.

Thanks

thinkyhead, 2018-05-24:
@Garyr14 — Please join the party at #10446

boelle, 2019-02-20:
@MarioStoilov do you have heatsinks mounted with thermal tape? do you have a fan blowing air at the drivers?

MarioStoilov, 2019-02-20:
Yes, a 120mm fan and heatsinks on all drivers

boelle, 2019-02-20:
did it solve this issue?

MarioStoilov, 2019-02-20:
No, it did not.

As I said, the extruder is geared (titan with a gear ratio 1:3) which means it's doing 3 times more work than a normal extruder would, so I think that is the main problem

boelle, 2019-02-20:
i have same extruder (from e3d) and a 80mm fan do work fine or me

what version are you running?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead